realsenseai/librealsense #2911: Cannot use Rs Align and Rs Point Cloud in RsProcessingPipe at the same time

yuka746, 2018-12-17:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  RealSense SDK 2.0                         | 
| Language                            |  unity                           | 
| Segment			| VR                          | 

### Issue Description

How can I use Rs Align and Rs Point Cloud of in RsProcessingPipe at same time?
I can use only Rs Align, but I activate Rs Point Cloud,  then Unity Editor stopped.

Please help me.

matkatz, 2018-12-18:
Hi,
The pointcloud filter align the frames internally.
The `PointCloudDepthAndColor` sample demonstrates how to do that using `PointCloudDepthAndColor`
processing pipe.
[image]

yuka746, 2018-12-19:
Thanks matkatz.
but I wanna show PointCloud and calculate position of object in color image.
so I have to use aligned depth image.

matkatz, 2018-12-19:
I added pointcloud filter on the the `AlignmentSample` processing pipe and the editor is streaming with no issues. (tested on v2.17.0)
There is an alignment issue that I need to check in case the pointcloud filter comes before the align filter in the pipe.
You can see my configuration below, please update to v2.17.0 in case you are using an older version and let me know if you still see this issue.
[image]

closed 2018-12-20 by RealSense-Customer-Engineering