realsenseai/librealsense #2912: Change resolution of images using matlab sdk

msalamah, 2018-12-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 }                                                   | 
| Firmware Version                   | (05.10.03.00)                                           | 
| Operating System & Version |   {Win 10                                                 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                          | 
| Platform                                 | PC                                                           |
| SDK Version                            |  {2.0 }                          | 
| Language                            | matlab }                          | 
| Segment			|  {others }                          | 

### Issue Description
i'm working with the matlab sdk, and i want to control the images resolution (both the depth and the rgb). according to the depth example the default resolution of the depth maps 1280x720 and the rgb resolution 640x480, i want to take a smaller images, from the viewer tool there are a lot of resolutions to choose.
how can i change the resolution using matlab sdk ?
from the images i can see that the depth images wider than the rgb image, how can i make the two images matching the same view ?

dorodnic, 2018-12-17:
Hi @msalamah 
Unfortunately, we still don't have many Matlab examples, but I recommend taking a look at [native examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras). Matlab API is more or less one-to-one translation. 
So, to configure resolutions, you want to create `realsense.config` object and use `enable_stream` functions. 
If you want the images aligned, you want to create `realsense.align` object and use `process` function (that takes the frameset from pipeline and outputs aligned frameset).
Let us know if you have more questions.

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Closing it now. Reopen if issue was not addressed.

closed 2019-01-10 by RealSense-Customer-Engineering

inofhand, 2019-05-11:
I need to adjust the resolution of the depth camera from 1280*720 to 640*320. How do I use the (enable_stream) command?

inofhand, 2019-05-11:
I am using D435 for depth photography, but I need to get depth information (10mm) from a closer distance. So I need to adjust the resolution of the depth camera from 1280*720 to 640*320. How do I use the (enable_stream) command to adjust the resolution? Can you give an example?

inofhand, 2019-05-11:
I am using D435 for depth photography, but I need to get depth information from a closer distance (100mm). So I need to adjust the resolution of the depth camera from 1280*720 to 640*320. How do I use the (enable_stream) command to adjust the resolution? Can you give an example?