realsenseai/librealsense #2913: Can't get temperature of motion module

UnaNancyOwen, 2018-12-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D435i)                                          | 
| Firmware Version                   | 05.10.13.00                                             | 
| Operating System & Version | Win 10 (1809)                                         | 
| Platform                                 | PC                                                           |
| SDK Version                            | 2.17.0                                                     | 
| Language                            |  C++                                                          | 

### Issue Description
I can't get temperature of motion module from RealSense D435i.
It displays an error "<code>query() motion_module_temperature_option failed! Empty buffer arrived.</code>".
What do you think?

```cpp
rs2::config config;
config.enable_stream( rs2_stream::RS2_STREAM_GYRO, rs2_format::RS2_FORMAT_MOTION_XYZ32F, 200 );
config.enable_stream( rs2_stream::RS2_STREAM_ACCEL, rs2_format::RS2_FORMAT_MOTION_XYZ32F, 63 );

rs2::pipeline pipeline;
rs2::pipeline_profile pipeline_profile = pipeline.start( config );

const std::vector<rs2::sensor> sensors = pipeline_profile.get_device().query_sensors();
for( const rs2::sensor& sensor : sensors ){
    if( sensor.supports( rs2_option::RS2_OPTION_MOTION_MODULE_TEMPERATURE ) ){
        const float temperature = sensor.get_option( rs2_option::RS2_OPTION_MOTION_MODULE_TEMPERATURE ); // error
        std::cout << "Motion Module Temperature : " << temperature << std::endl;
    }
}
```

dorodnic, 2018-12-17:
Hi @UnaNancyOwen 
We are still debating if motion-module temperature is something we want to expose. 
In previous D400+IMU previous product we used to have the IMU on a separate PCB to the right of the main board. With how things are configured right now, ASIC temperature should offer sufficiently good approximation. That said, we are going to address it in one way or another, it shouldn't throw an exception.

closed 2019-03-18 by RealSenseCustomerSupport

flipphillips, 2019-09-16:
Even though this is closed... just wanted to note that, in real sense viewer on macOS, built from source as of 16 Sept 2019, this error is still being thrown... even in developer branch.