realsenseai/librealsense #2914: Apply texture on filtered point cloud data

nfadeeva, 2018-12-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Win (8.1) on Parallels VM on MacOS  | 
| Platform                                 | PC |
| SDK Version                            |   2.16.5                          | 
| Language                            | C++                         | 

### Issue Description
The same issue as https://github.com/IntelRealSense/librealsense/issues/2100
I am trying to view the textured filtered point cloud data using the code below. The point cloud data does not have the colored texture on it. Could you please explain what is wrong with my code?

`#include <librealsense2/rs.hpp> 
#include "example.hpp"        

#include <algorithm>            
#include <ctime>
#include <map>         
int main(int argc, char * argv[])
{
	std::clock_t start;
	double duration;

	start = std::clock();

	if (argc != 4)
	{
		std::cout << "pointcloud.exe input.bag . time_btwn_frames_sec"
			<< std::endl;
		return 0;
	}

	

	std::string filename_bag = argv[1];
	std::string output_dir = argv[2];

	// min time between frames in sec
	float time_btwn_frames = std::atof(argv[3]);

	rs2::config cfg;
	cfg.enable_device_from_file(filename_bag, true);

	rs2::pipeline pipe;
	auto profile = pipe.start(cfg);
	auto device = pipe.get_active_profile().get_device();
	rs2::playback playback = device.as<rs2::playback>();
	playback.set_real_time(false);

	int c = 0;

	// Declare pointcloud object, for calculating pointclouds and texture mappings
	rs2::pointcloud pc;
	// We want the points object to be persistent so we can display the last cloud when a frame drops
	rs2::points points;

	auto frameNumber = 0ULL;
	float curTime = 0;
	float diff = 0;

	while (true) 
	{
		playback.resume();
		// Wait for the next set of frames from the camera
		auto frames = pipe.wait_for_frames();
		playback.pause();

		auto depth = frames.get_depth_frame();
		if (frames[0].get_frame_number() < frameNumber) {
			break;
		}
		float t = float(playback.get_position()) / (10000 * 100000);
		if (t - curTime < time_btwn_frames)
		{
			continue;
		}
		curTime = t;
		//std::cout << curTime << std::endl;
		frameNumber = frames[0].get_frame_number();

		// Generate the pointcloud and texture mappings
		

		// Tell pointcloud object to map to this color frame
		//pc.map_to(color);
		c += 1;
		if (c == 1)
			continue;
		

		rs2::decimation_filter dec_filter;  // Decimation - reduces depth frame density

		// Initialize a vector that holds filters and their options

		rs2::frame filtered = depth;
		//filtered = depth_to_disparity.process(filtered);
		filtered = dec_filter.process(filtered);
		

		points = pc.calculate(filtered); // Generate pointcloud from the depth data
		auto color = frames.get_color_frame();
		if (!color)
			continue;
		pc.map_to(color);

		std::string name = output_dir + "/" + std::to_string(c) + ".ply";

		points.export_to_ply(name, color);
	}
	duration = (std::clock() - start) / (double)CLOCKS_PER_SEC;

	std::cout << "printf: " << duration << '\n';

	return EXIT_SUCCESS;
}`

RealSense-Customer-Engineering, 2018-12-20:
[Realsense Customer Engineering Team Comment]
@nfadeeva Noticed that you're trying to read from *.bag file. How do you get this *.bag file? Please make sure the color streaming is enabled when recording to *.bag file if you want to see colored texture.

nfadeeva, 2018-12-20:
> [Realsense Customer Engineering Team Comment]
> @nfadeeva Noticed that you're trying to read from *.bag file. How do you get this *.bag file? Please make sure the color streaming is enabled when recording to *.bag file if you want to see colored texture.

This .bag file is the record from the sensor d415, the main point of this issue is the fact that without a filter (points = pc.calculate(**depth**)) I a get color cloud but when I apply decimation (points = pc.calculate(**filtered**);) - the cloud is totally black.

RealSense-Customer-Engineering, 2018-12-24:
[Realsense Customer Engineering Team Comment]
@nfadeeva  Please configure playback.set_real_time to "true" to try again.

    playback.set_real_time(true);

RealSense-Customer-Engineering, 2018-12-24:
[Realsense Customer Engineering Team Comment]
@nfadeeva  And another point is that it costs time to save *.ply.  You can have a try with a short *.bag file to see if it's working.

nfadeeva, 2018-12-25:
> [Realsense Customer Engineering Team Comment]
> @nfadeeva Please configure playback.set_real_time to "true" to try again.
> 
> ```
> playback.set_real_time(true);
> ```

I've changed this row but the cloud is still totally black

nfadeeva, 2018-12-25:
> [Realsense Customer Engineering Team Comment]
> @nfadeeva Please configure playback.set_real_time to "true" to try again.
> 
> ```
> playback.set_real_time(true);
> ```

and also if I try to stream from the real sensor (not read from .bag) result is the same - totally black or gray clouds

nfadeeva, 2018-12-26:
> [Realsense Customer Engineering Team Comment]
> @nfadeeva Seen from the result I run here, your code will generate several *.ply files. The first ply file is black but the later files should have color.
> BTW, the librealsense version I used is 2.17.0 and FW 5.10.6. The bag file was captured from realsenseviewer with depth and color 1280x720@30fps. And the total duration is 2 seconds.
> Attached is a snapshot for 38.ply.

I've installed 2.17.0 and clouds have color after decimation, thank you, the issue is solved

closed 2018-12-26 by nfadeeva