realsenseai/librealsense #2916: Low RGB fps pulls down Depth framerate

joshmer, 2018-12-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.9.13 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.17.0  | 
| Language                            |  C#  | 

### Issue Description
Using the latest, official Intel driver, I've found that depth/RGB framerates are somehow linked together. For a given Depth stream framerate, I *must* select the same framerate for the RGB stream otherwise the Depth framerate drops.

If I use the default Microsoft driver, I don't have this issue. It only occurs with the official Intel driver.

For example. let's say I want to stream Depth data at 30fps and RGB data at 6fps. When my application runs, the RGB stream shows 6fps however the Depth stream is only working at 12fps. If I bump the RGB stream to 15fps, then Depth will show ~18-20fps. The only way I can get the Depth stream to show a full 30fps is to also set the RGB stream to 30fps.

Has anyone seen this behavior before?

RealSense-Customer-Engineering, 2018-12-20:
[Realsense Customer Engineering Team Comment]
Hi @joshmer,

Which camera do you use? D415 or D435 camera?
Did you try realsense-viewer and also see the issue?
Per the depth frame number and color frame number, it looks fine based on depth 30FPS and RGB 6FPS.
the sample code like:
depth.get_frame_number()
color.get_frame_number()

joshmer, 2019-01-02:
Hi @RealSense-Customer-Engineering 

Sorry for the delay!

For this specific configuration, we use a D415, but I tested with a D435 and found the same results with the Intel driver.

I was also able to confirm this behavior via the realsense-viewer v2.16.5 with firmware v5.09.13. If I only run the depth camera at 30fps, everything is smooth. When I enable the RGB camera at 6fps, realsense-viewer reports 30fps for depth but it's immediately dropping frames. The RGB stream shows 0fps and randomly shows a 'No frames received' message.

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

Hi @joshmer,

Would you like to have a try of the latest librealsense and firmware and see how it goes on your system setup?

RealSenseCustomerSupport, 2019-03-06:
----------------------------------------------

Hi @joshmer,

Any test update based on the latest librealsense and firmware?

joshmer, 2019-03-10:
@RealSenseCustomerSupport 

Apologies for the delay! I haven't had a chance to test the updated drive yet. I will update you ASAP.

closed 2019-05-09 by RealSenseCustomerSupport