realsenseai/librealsense #2918: rs-save-to-disk how to specify location for saved images?

moranbar, 2018-12-18:
| Camera Model                         D435
| Operating System & Version   Linux Ubuntu 16
| SDK Version                            2

im using rs-save-to-disk,
 how to specify location for saved images?

i cant even find the ones i already took.

much appriciated.

ev-mp, 2018-12-18:
Hell @moranbar,
The `rs-save-to-disk` outputs are saved into the current directory ('pwd').
This example's focus is to demonstrate an use-case and API usage, hence only the  functionalities essential to the main success scenario are present. Though it can always be improved via enhancement PR.

RealSense-Customer-Engineering, 2018-12-22:
[Realsense Customer Engineering Team Comment]
Hi @moranbar,

Still need any support for this topic? You can also modify the code and share to everyone for the improvement.

closed 2019-02-19 by RealSenseCustomerSupport