realsenseai/librealsense #2922: Need GetSensorTime API

bentank, 2018-12-18:
Required InfoCamera ModelD435Firmware Version05.09.14.00Operating System & VersionWin 10PlatformPCSDK Version2.11.1Question about timestampsI have a question about how the timestamps that are present in the metadata are generated. I'm using Nuitrack to extract skeletons from the image data and I need to synchronize this data with other devices. My main question is: to what event do the timestamps belong and what is the clock from which they are generated? My guess so far is:SENSOR_TIMESTAMP: Device clock / sensor starts taking the imageFRAME_TIMESTAMP: Device clock / frame starts to transfer to the driverBACKEND_TIMESTAMP: PC clock / frame starts to transfer to the driverTIME_OF_ARRIVAL: PC clock / frame is transfered to the driverHopefully you can give me more clarity on this, I couldn't find it anywhere. | Required Info |   | Camera Model | D435 | Firmware Version | 05.09.14.00 | Operating System & Version | Win 10 | Platform | PC | SDK Version | 2.11.1
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

Camera Model | D435
Firmware Version | 05.09.14.00
Operating System & Version | Win 10
Platform | PC
SDK Version | 2.11.1

Required Info |  
-- | --
Camera Model | D435
Firmware Version | 05.09.14.00
Operating System & Version | Win 10
Platform | PC
SDK Version | 2.11.1

This is a follow up to the closed issue in #2188.

Please provide a reasonable response without just closing out the issue. 

Thanks!

bentank, 2018-12-18:
@dorodnic, thanks for the update. Please let me know if I can assist with more information regarding the issue and purposed solution. It is greatly appreciated!

dorodnic, 2018-12-19:
Hi @bentank 

Good news! While there is no such "official" command in firmware, you can read the value off the relevant register directly, and this works on production units (on NDA hardware you could also **write** registers, but this is blocked on production units for obvious reasons)

Basically, you can use the following code snippet to get current firmware time:
```cpp
using namespace rs2;

pipeline p;
auto prof = p.start();
auto dev = prof.get_device().as<debug_protocol>();
uint8_t data[]{
    0x14, 00, 0xab, 0xcd, 01, 00, 00, 00, 0x3c, 0x61, 01, 00, 0x40, 0x61, 01, 00, 00, 00, 00, 00, 00, 00, 00, 00
};
std::vector<uint8_t> command(data, data + sizeof(data));
auto res = dev.send_and_receive_raw_data(command);
auto ptr = reinterpret_cast<uint32_t*>(res.data());
auto ts = ptr[1] * 0.001;
std::cout << ts << std::endl;
```

There is a concern of how much time such query can take. In theory it goes through the entire UVC driver stack, and can be bounced by the firmware several times (because it relies on an asynchronous UVC extension unit), but in practice I'm getting a result within ~1ms. This should be good enough for any practical purpose.

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Closing it for now. Please reopen if the issue was not addressed.

dorodnic, 2019-05-08:
@bentank - #3909 is the follow-up on this request, if you are interested.

dorodnic, 2019-06-03:
With "Global Time" part of the 2.22 release and working well, I think it's safe to close this one :)

closed 2019-06-03 by dorodnic