realsenseai/librealsense #2923: Problem with using Qt creator

tarekmuallim, 2018-12-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   |   | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.5                           | 
| Language                            |  C++                          | 
| Segment			|                           | 

Hi

I am trying to develop an app using RealSense D415 sensor.
The software is using C++ and Qt creator and the operating system is windows 10. But on the build, I got an error.
It looks that there is some conflict between RealSense SDK and a macro in Qt “Q_FOREACH”.

How can I solve the problem without rebuilding RealSense SDK?
I upload a simple Qt project for test purposes.

Thanks.

[testd415.zip](https://github.com/IntelRealSense/librealsense/files/2693487/testd415.zip)

SimonLit, 2018-12-19:
Change the `foreach` in `..\Intel RealSense SDK 2.0\include\librealsense2\hpp\rs_frame.hpp` to a different name, par example to `foreachRS,`. No need to rebuild the SDK.

tarekmuallim, 2018-12-19:
@SimonLit I will try it.
Thank you.

tarekmuallim, 2018-12-20:
It work perfectly.
Thank you.

closed 2018-12-21 by RealSense-Customer-Engineering

Pabebezz, 2018-12-22:
> ``hi，I also met this problom. also change "foreach " to"foreach_rs" in rs_frame.hpp .But the program is broken. Can you tell me what's your complier? mingw or mvcs?   
    other problem  May my code has any problem? Can you help me ?
Thank you!

 int main(int argc, char * argv[])
{
    QApplication a(argc, argv);
    MainWindow w;
    pipeline pipe;
    config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR,640,480,RS2_FORMAT_BGR8,30);
    pipe.start(cfg);
    frameset frames;
    for(int i = 0; i < 30; i++ ){

        frames = pipe.wait_for_frames();
    }
    frame color_frame = frames.get_color_frame();
    Mat color(Size(640,480),CV_8UC3,(void*)color_frame.get_data(),Mat::AUTO_STEP);
    imshow("realimage",color);
    w.show();
    return a.exec();
}

tarekmuallim, 2018-12-24:
@Pabebezz
I am using MinGw 7.3.0 64bit
My operating system is windows 10 64bit
I linked the x64 realsense library to my Qt project.

tarekmuallim, 2018-12-26:
@Pabebezz
In you project configuration
Keep only the first include path
The rest are not needed. You don't have to include the subdirectories. 
For the library delete the - L sembol from the beginning of the path and
Add only .lib file but for x64 . 
Also add the sdk bin directory to the system path.

Pabebezz, 2018-12-26:
@tarekmuallim  Yep ,It work !   Thank you!