realsenseai/librealsense #2924: usbids conflicts with Movidius Neural Compute Stick

yechun1, 2018-12-19:
Issue: The usb ids of idProduct and idProduct conflict with Movidius NCS.
Reproduce:
1.plugin NCS

> [ 4813.574892] usb 1-3: USB disconnect, device number 44
> [ 4813.940359] usb 1-3: new low-speed USB device number 46 using xhci_hcd
> [ 4814.097708] usb 1-3: New USB device found, idVendor=046a, idProduct=0011
> [ 4814.097713] usb 1-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
> [ 4814.108901] input: HID 046a:0011 as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:046A:0011.000F/input/input32
> [ 4814.169124] hid-generic 0003:046A:0011.000F: input,hidraw0: USB HID v1.11 Keyboard [HID 046a:0011] on usb-0000:00:14.0-3/input0
> [ 4814.440332] usb 1-4: new high-speed USB device number 47 using xhci_hcd
> [ 4814.588842] usb 1-4: New USB device found, idVendor=03e7, idProduct=2150
> [ 4814.588848] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 4814.588853] usb 1-4: Product: Movidius MA2X5X
> [ 4814.588856] usb 1-4: Manufacturer: Movidius Ltd.
> [ 4814.588860] usb 1-4: SerialNumber: 03e72150

2.plugin Realsense

> [ 4799.600384] usb 2-1: new SuperSpeed USB device number 17 using xhci_hcd
> [ 4799.621106] usb 2-1: New USB device found, idVendor=8086, idProduct=0b07
> [ 4799.621111] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 4799.621114] usb 2-1: Product: Intel(R) RealSense(TM) 435
> [ 4799.621118] usb 2-1: Manufacturer: Intel(R) RealSense(TM) 435
> [ 4799.621122] usb 2-1: SerialNumber: 821413020849
> [ 4799.622191] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
> [ 4799.622200] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
> [ 4799.622321] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
> [ 4799.624007] uvcvideo: Unable to create debugfs 2-17 directory.
> [ 4799.624132] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
> [ 4799.624136] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
> [ 4799.624140] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) 435 with was not initialized!
> [ 4799.624144] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
> [ 4799.624392] input: Intel(R) RealSense(TM) 435: Int as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input31
> [ 4799.625083] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
> [ 4799.625095] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 435 (8086:0b07)
> [ 4799.625879] uvcvideo: Unable to create debugfs 2-17 directory.
> [ 4799.625987] uvcvideo 2-1:1.3: Entity type for entity Processing 7 was not initialized!
> [ 4799.625993] uvcvideo 2-1:1.3: Entity type for entity Extension 8 was not initialized!
> [ 4799.625997] uvcvideo 2-1:1.3: Entity type for entity Camera 6 was not initialized!

3.run realsense-viewer
4.the NCS device missed (lsusb)

Realsense rules:
https://github.com/IntelRealSense/librealsense/blob/c5b9aa0127bb4dcabb35275360ca51e5c53da693/config/99-realsense-libusb.rules#L27

NCSSDK rules:
https://github.com/movidius/ncsdk/blob/949131b4402c4ae4b7d1c7d11bb4ac9e70d703da/api/src/97-usbboot.rules#L1

Official defined here:
http://www.linux-usb.org/usb.ids
03e7  Intel
	2150  Myriad VPU [Movidius Neural Compute Stick]

RealSense-Customer-Engineering, 2018-12-26:
[Realsense Customer Engineering Team Comment]
Hi @yechun1,

I don't see this issue. If you re-connect NCS1, can it be recognized and show in lsusb?
You can try to connect NCS1 to another USB port, not near D435 device. I wonder if any interference there.

RealSense-Customer-Engineering, 2018-12-27:
[Realsense Customer Engineering Team Comment]
Hi @yechun1,

Did you run the prebuilt realsense-viewer tool? Or, build from the source?
If you build from the source, is "BUILD_WITH_TM2" build option enabled?

I suspect the NCS1 is reset by realsense-viewer since our TM2 device has the same VID/PID with NCS1.
That's the reason you saw NCS1 is disconnected. My suggestion is to build realsense-viewer tool without "BUILD_WITH_TM2".
Let me know how it goes.

yechun1, 2018-12-27:
Just verified, NCS1 could work without BUILD_WITH_TM2.

The issue was reported from ros2debian, the debian package was prebuilt by ROS2 buildfarm based default CMakeList.txt. Don't know why the value of BUILD_WITH_TM2 is true by default? As there nowhere set it from CMakelist.txt. I have submitted a PR(https://github.com/IntelRealSense/librealsense/pull/2970) to fix this issue.

Disable BUILD_WITH_TM2 seems to be a workaround solution, is there any better way to fix the same VID/PID issue?

Anyway, it can solve our currently problem as we do not use TM2, thanks.

closed 2019-02-19 by RealSenseCustomerSupport