realsenseai/librealsense #2925: How to save depth image as 4000 x depth in meter in a 16bit PNG

SunshineDou, 2018-12-19:
Hello, as I want to saved aligned depth image and rgb image, my code is as follows, but I want to save the depth image as 4000 x depth in meter in a 16bit PNG, How to do it using python?

import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

decimation = rs.decimation_filter()
spatial = rs.spatial_filter()
temporal = rs.temporal_filter()
hole_filling = rs.hole_filling_filter()
depth_to_disparity = rs.disparity_transform(True)
disparity_to_depth = rs.disparity_transform(False)

profile = pipeline.start(config)

depth_sensor = profile.get_device().first_depth_sensor()

preset_range = depth_sensor.get_option_range(rs.option.visual_preset)
print('preset range:'+str(preset_range))
for i in range(int(preset_range.max)):
        visualpreset=depth_sensor.get_option_value_description(rs.option.visual_preset,i)
if visualpreset == "custom":
        depth_sensor.set_option(rs.option.visual_preset, i)

if depth_sensor.supports(rs.option.depth_units):
	depth_sensor.set_option(rs.option.depth_units,0.0001)

depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

clipping_distance_in_meters = 4 #1 meter
clipping_distance = clipping_distance_in_meters / depth_scale
align_to = rs.stream.color
align = rs.align(align_to)

i=1
j=1
try:
    while True:
        frames = pipeline.wait_for_frames()
        
        aligned_frames = align.process(frames)
        
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()
        
        if not aligned_depth_frame or not color_frame:
            continue
        
        frame = decimation.process(aligned_depth_frame) 
        frame = depth_to_disparity.process(aligned_depth_frame)
        frame = spatial.process(frame)
        frame = temporal.process(frame)
        filtered_depth_frame = disparity_to_depth.process(frame)
        #filtered_depth_frame = hole_filling.process(frame)

        depth_image = np.asanyarray(filtered_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        
        grey_color = 153
        depth_image_3d = np.dstack((depth_image,depth_image,depth_image)) #depth image is 1 channel, color is 3 channels
        bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)
        
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((bg_removed, depth_colormap))
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        cv2.imwrite("//home//cindy//401TestLighting//depth//"+str(i)+".png",depth_image) 
        cv2.imwrite("//home//cindy//401TestLighting//rgb//"+str(i)+".png",color_image) 
        i = i+1
        j = j+1
        key = cv2.waitKey(1)
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()

Thanks very much

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Hi SunshineDou,

Wonder what do you mean by "4000 x depth" in your question?
>>>
I want to save the depth image as 4000 x depth in meter in a 16bit PNG
<<<

Thanks!

RealSenseCustomerSupport, 2019-01-21:
Hi SunshineDou,

Wonder if there is any update?

Thanks!

RealSenseCustomerSupport, 2019-01-28:
Hi SunshineDou,

If there is nothing else, this one will be closed soon.

Thanks!

closed 2019-02-06 by RealSenseCustomerSupport