realsenseai/librealsense #2926: D415 URDF and Mesh for ROS/MoveIt

tragolf, 2018-12-19:
Hi,

I have been using the D435 with ROS. For this model I could find correct URDF-Files and Mesh and it works fine. Now I want to use the D415 but I can't find the URDF and Mesh. I can add the camera as a blackbox to the endeffector of my robot but I don't now the internal transformations between the cam, ir, depth etc frames. Therefore my detections of objects are off.
Does anyone know where to get these files? Especially the internal tf between cam left, cam right, depth... frames is important.
Thanks

dorodnic, 2018-12-24:
You can get the STEP file [here](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel_RealSense_415_0601_EXT-REV-02_25_MAY_2017.STEP).

This can be converted to OBJ and other 3D formats using freecad on Linux. 

We will try to add the URDF as soon as possible.

closed 2019-01-11 by RealSense-Customer-Engineering

tragolf, 2019-01-14:
Is the URDF for D415 already available (the issue was closed)? Where can I find the URDF for d415? In the official repo it has not been added yet. Thanks for the support