realsenseai/librealsense #2927: D435 for reconstruction

SunshineDou, 2018-12-19:
Hello, I want to reconstruct a room by Kintinuous(an open source 3D reconstruction project), I want to use D435, is it OK? how far should I take the datasets(aligned depth and rgb images)? 
Thanks very much

HippoEug, 2019-01-07:
Hi @SunshineDou , did you find anything regarding object/area reconstruction?

RealSense-Customer-Engineering, 2019-01-09:
[Realsense Customer Engineering Team Comment]
Hi SunshineDou,

There is a really good document for 3D reconstruction sample using Kinect Fusion:
The sample should works with D435. Please have a quick look. Thank you.

SunshineDou, 2019-01-09:
@HippoEug I find the reconstruction result is not very good and it drift noticeably as I said in#3031

SunshineDou, 2019-01-09:
@RealSense-Customer-Engineering Thank you very much for your replying. I will have a look at it soon

HippoEug, 2019-01-09:
I am a little confused about Kinect Fusion as mentioned by @RealSense-Customer-Engineering . Is there a link to download it? Or is it just another name for RecFusion?

closed 2019-01-16 by RealSense-Customer-Engineering