realsenseai/librealsense #2930: How to get the fx,fy,cx,cy of the D435

SunshineDou, 2018-12-19:
hello, as I know that RealSense cameras are usually well calibrated when purchased, and I bought realsense D435, I need to know the intrinsic including fx,fy,cx,cy of the camera, how can I get it ?
Thank you very much

jSoftSys, 2018-12-19:
rs2::pipeline		pipe;
rs2::pipeline_profile	pipeProfile = pipe.start(_cfg);

pipeProfile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>().get_intrinsics()

SunshineDou, 2018-12-19:
@jSoftSys Thank you very much for your replying. I am very sorry that I want to ask can I get it using python? and in the above code, I can only get py, py, fx, fy, how can I get cx and cy?
Thank you very much for your help sincerely

lramati, 2018-12-19:
The code is very similar as the python API is 1:1 with the C++ API.

The following should work:

```python
import pyrealsense2 as rs
pipe = rs.pipeline
pipe_profile = pipe.start()

pipe_profile.get_stream(rs.stream.depth).as_video_stream_profile().get_intrinsics()
```

SunshineDou, 2018-12-19:
@jSoftSys Thanks for your helping. I write it as follows:
`config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

decimation = rs.decimation_filter()
spatial = rs.spatial_filter()
temporal = rs.temporal_filter()
hole_filling = rs.hole_filling_filter()
depth_to_disparity = rs.disparity_transform(True)
disparity_to_depth = rs.disparity_transform(False)

profile = pipeline.start(config)

intr = profile.get_stream(rs.stream.depth).as_video_stream_profile().get_intrinsics()
rs.set(intrinsics.Width,480)
rs.set(intrinsics.Height,640)
print("ppx is: ",intr.ppx)
print("ppy is: ",intr.ppy)
print("fx is: ",intr.fx)
print("fy is: ",intr.fy)
print("cx is: ",intr.cx)
print("cy is: ",intr.cy)`

but it report errors on"rs.set(intrinsics.Width,480)", but if I don't set it , the output ppx is  322.7760009765625, ppy is 239.79263305664062, but I want to get the 640x480's ppx and ppy, how to set it?
The other question is that how to get  cx and cy? do I need to calculate it? as I am a beginner in this field. thanks very much

jSoftSys, 2018-12-19:
The cameras principle point are descried by ppx and ppy, thus correspond to cx and cy respectively which is usally used in literature..

SunshineDou, 2018-12-20:
@jSoftSys Oh, I get it now. Thank you very much for your generous help

closed 2018-12-20 by SunshineDou