realsenseai/librealsense #2931: Question: convert the RealSense SDK's Unity wrapper to C# Jobs and ECS in future?

MartyG-RealSense, 2018-12-19:
Hi,

In Unity 2018.2, a new processing technology called Jobs was introduced and ECS (Entity Component System) were introduced.  It works by processing data inside 'structs' instead of classes so that processing tasks can be divided up between the multiple cores of a processor chip.  

ECS can additionally be used to create speedy 'burst compilation'.

The optimization achieved by these systems is said to give a speed increase of up to 50%.  An example was shown where around 17,000 spacecraft could be onscreen under the old system before the FPS dropped below 30 FPS, whilst around 25,000 spacecraft could be onscreen under the new system before FPS dipped below 30.

Unity provide an example program where a point cloud can be generated and processed with burst compilation.

https://github.com/stella3d/job-system-cookbook/blob/master/Assets/Scripts/PointCloudProcessing.cs

Unity see the Job system and ECS as being the future of Unity, switching from object oriented programming to Data-oriented programming.  

So my question is: might parts of the code of the RealSense SDK's Unity wrapper be convertable in future to Jobs and ECS in order to provide boosted performance in areas such as point cloud processing?

ogoshen, 2018-12-31:
Those examples are for **generating** dynamic point cloud / mesh data in an optimized way, which the sdk already does.

But suppose you wanted to manipulate those incoming point clouds per frame, you could definitely benefit using those features.

The `RsPointCloudRenderer` does have overhead, when updating the mesh, memory is actually copied 3 times ([here](https://github.com/IntelRealSense/librealsense/blob/b7eef8c2a2ad9a7f4a62335f27da40b0cc8b80da/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsPointCloudRenderer.cs#L186)):
1. Native to managed
2. Managed to mesh vertices
3. Vertices to GPU

This can be circumvented by writing a [native rendering plugin for Unity](https://docs.unity3d.com/Manual/NativePluginInterface.html) that uploads native data straight to GPU.

I wouldn't go there so quickly though, performance is pretty good as it is and writing such a plugin with all the threading issues and graphics API differences can be difficult.

Since Unity 2018 there is another, more portable way (albeit unsafe):
1. Convert [native data to NativeArray](https://docs.unity3d.com/ScriptReference/Unity.Collections.LowLevel.Unsafe.NativeArrayUnsafeUtility.ConvertExistingDataToNativeArray.html) 
2. Use the (undocumented) unsafe `NativeArray<T>` overload of [ComputeBuffer.SetData](https://docs.unity3d.com/ScriptReference/ComputeBuffer.SetData.html)
3. Finally do a procedural draw call with [Graphics.DrawProcedural](https://docs.unity3d.com/ScriptReference/Graphics.DrawProcedural.html)

I imagine this will get even easier now that Unity officially moves to .NET 4, I expect to see more APIs using `NativeArray<T>` and `Span<T>`

MartyG-RealSense, 2018-12-31:
@ogoshen  Thanks so much for your input on my question!  Converting thinking from the traditional MonoBehaviour script layout to Structs to take advantage of the Job system's multi-core processing in my own project has certainly been challenging, but I'm starting to get the hang of it now.  I'm using the NativeArray in my project.

I'm particularly keen to see the advantages it has for use with Unity's next-generation online multiplayer tech, which can connect to Multiplay dedicated servers (the Multiplay server company was purchased by Unity).  It would be cool to have multiple avatars with RealSense camera controls all together in a shared online environment.  At present, I have the basic client-server connection code up and running, so I look forward to being able to set up such an environment.  *smiles*

closed 2019-01-14 by RealSense-Customer-Engineering