realsenseai/librealsense #2932: Plugged in D435 at boot gives errors

kevindehecker, 2018-12-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.10.06.00 | 
| Operating System & Version |   Linux Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.13.0-43-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |   2.17.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I get the following error in the Realsense viewer:
[image]
```
Invalid Value in rs2_get_option(options:0x7ffb4c005850, option:Emitter Enabled):
get_xu(id=2) failed! Last Error: Input/output error
```

...When I do the following:

1. Power down pc
2. Plug in RealSense
3. Power up, boot pc, and start realsense-viewer
4. Change a setting, such disable the Emitter. This will result in the error.

Step 4 can also be reproduced from a skeleton code project I have, which does nothing else then enumerating the RealSense device, and attempt to change a setting (although in this case, I have to switch the same setting on _and_ off before some similar error occurs). I can reproduce this issue on all my systems (various types of x86 linux systems).

Currently, I can circumvent the problem in two ways:
1a. Replug the usb cable. This only has to be done once per boot, then this problem never happens anymore. 
1b. Don't boot the system with the RS already plugged in
2. Perform a hardware reset in the rs2::device. Although this does seem to fix the problem, doing this preventively seems to give a set of problems of its own. (e.g. I got a access violation error )

haydenhager, 2019-02-05:
I am also experiencing this issue. It seems to occur for me when after a sleep state. I have to unplug and replug to remedy. 

64-Bit Ubuntu 16.04 LTS, realsense viewer V2.17.1, Intel D415 FW 05.10.13.00. 

This occurs in the viewer and programmatically connecting through python script.

kevindehecker, 2019-02-10:
The problem still is there with 2.18.1 and FW 05.11.01.00

RealSenseCustomerSupport, 2019-02-11:
----------------------------------------------

@kevindehecker Yes, this problem is still under triage. Will update here when get fix. Thanks for your patience!

ev-mp, 2019-02-11:
Hello @kevindehecker,
Please collect and attach the firmware log for review:
1. Run the realsense-viewer or any other executable that triggers the error.
2. Run FW log collection: 
>rs-fw-logger.exe  > fwlog.log

Also, is it possible that you're using a pre-production unit ? (camera's serial number can be useful as well)

TheMikeyR, 2019-02-11:
I'm having the same issue with this see screenshot. Happens both to D435 and D415

Using v2.18.0
FW: 05.11.01.00
not sure what is s/n but here is what is on the sticker of the D435
J72479-100
817612070517

[image]

[fwlog.log](https://github.com/IntelRealSense/librealsense/files/2850317/fwlog.log)

[librealsense.log](https://github.com/IntelRealSense/librealsense/files/2850318/librealsense.log)

ev-mp, 2019-02-11:
@TheMikeyR , the error you encounter is an unrelated regression in V2.18.0 that was fixed in 2.18.1.
(#3137, #3146, #3191, #3212).

TheMikeyR, 2019-02-11:
@ev-mp you are right it is fixed in 2.18.1

kevindehecker, 2019-02-11:
@ev-mp 
[fwlog.log](https://github.com/IntelRealSense/librealsense/files/2853177/fwlog.log)

BTW, the rs-fw-logger does not exit, so I just ctrl-c-ed it after a minute orso.

I can reproduce with multiple D435's, which should not be pre-product. In any case the D435i which I had at hand and used for this test has serial:
k38179-100
838212070042

ev-mp, 2019-02-12:
Thanks for the input but the log is missing the device initialization part that is essential:
 - Please make sure to run the FW logger first, then the viewer so that the initial log messages will not be lost.
 - Also repeat this procedure with the D435 that you referred in the beginning of the thread.

The logger is not supposed to exit, so "ctrl+c"-ing is ok.

kevindehecker, 2019-02-12:
[loggielogD435.log](https://github.com/IntelRealSense/librealsense/files/2858017/loggielogD435.log)
[loggielogD435i.log](https://github.com/IntelRealSense/librealsense/files/2858018/loggielogD435i.log)

ev-mp, 2019-02-13:
@kevindehecker , thanks again.
The logs will be reviewed.

In order to facilitate the reproduction - 
Are you connecting the device to on-board USB ports directly or using an PCI-extension board? 
And if so - can you specify the make/model of the HW ?

kevindehecker, 2019-02-13:
No, this is directly connected to my laptop. Can also reproduce with a NUC 7i3BNK (again directly plugged in) and other laptops.

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

@kevindehecker Please try the latest FW v5.11.6.200 which should fix the issue. The FW can be got from
https://downloadcenter.intel.com/download/27954/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@kevindehecker Did you get chance to try FW v5.11.6.200? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@kevindehecker Didn't get response from you for several weeks. Will close this ticket at this point. Please feel free to open new tickets if you still have question, Thanks!

kevindehecker, 2019-05-20:
Sorry, was on vacation.  Will try it out Friday!

closed 2019-05-20 by RealSenseCustomerSupport

kevindehecker, 2019-05-24:
Hmm, well unfortunately I can still reproduce this problem with steps posted in the opening post.

But, I could only update to 5.11.4, the link you provided does not show v5.11.6.200 ...?

kevindehecker, 2019-05-27:
Can you reopen this please? Or should I go through the trouble of copying the whole thing into a new issue...?

sherrysun2018, 2019-06-06:
@kevindehecker  Sorry to know that you still have the issue. Please copy it to new issue and we will look into it.