realsenseai/librealsense #2934: Frames can't arrive - problem in Linux with 2.17 update

softyara, 2018-12-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Linux (Ubuntu 16 and 18)  | 
| Kernel Version (Linux Only)    |  4.15.0-42-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.17.0                          | 
| Segment			|  Robot                        | 

### Issue Description
With the 2.17.0 update can't receive any stream in realsense-viewer. Depth, Infrared, RGB, tried many different parameters - frames can't arrive and I just see white window for each stream. Same for both Ubuntu 16 and 18. Tried multiple D415 and D435 cameras. I use USB 3.
However, everything works if I install librealsense 2.16 or 2.15.

dorodnic, 2018-12-19:
Hi @softyara 
Thank you for reporting the issue. This is not something we have seen before with the 2.17 release.
Could you please share more information, such as CPU type, any warnings in the dmesg logs, if you are building from source or installing via apt-get? Did you install the librealsense-dkms package and does it make a difference?

softyara, 2018-12-20:
Hi @dorodnic, thank you for the answer. Seems like I found the cause of the problem: the cable and/or the usb port. At least, it started working properly with 2.17 release. It is still a little strange that it functioned with other versions of the lib...
I apologize for the false alarm.
I am closing this issue.

closed 2018-12-20 by softyara