realsenseai/librealsense #2935: SR300: /camera/driver - Error calling rs_wait_for_frames

pl95, 2018-12-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Operating System & Version |   Ubuntu 14.04 | 
| Kernel Version (Linux Only)    |  4.4.0                                        | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I installed the librealsense following the instruction in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#Troubleshooting

When I run
 **_roslaunch realsense_camera sr300_nodelet_default.launch_**
I have the following errors.

... logging to /home/fetch/.ros/log/48698a5a-03d0-11e9-9c86-d8cb8a5dca62/roslaunch-fetch6-4200.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fetch6:35116/

SUMMARY
========

PARAMETERS
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: SR300
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_height: 480
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/depth_width: 640
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/mode: manual
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /camera/
    driver (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera/nodelet_manager-1]: started with pid [4222]
process[camera/driver-2]: started with pid [4223]
[ INFO] [1545256733.842314026]: Initializing nodelet with 4 worker threads.
[ INFO] [1545256734.206235227]: /camera/driver - Detected the following camera:
				- Serial No: 620202003098, USB Port ID: 7-1-1, Name: Intel RealSense SR300, Camera FW: 3.17.0.0
[ WARN] [1545256734.206276330]: /camera/driver - Detected unvalidated firmware:
				- 620202003098's current camera firmware is 3.17.0.0, Validated camera firmware is 3.10.10.0
[ INFO] [1545256734.206292011]: /camera/driver - Connecting to camera with Serial No: 620202003098, USB Port ID: 7-1-1
[ INFO] [1545256735.584291085]: /camera/driver - Setting static camera options
[ INFO] [1545256735.585747249]: /camera/driver - Enabling Depth in manual mode
[ INFO] [1545256735.586014152]: /camera/driver - Enabling Color in manual mode
[ INFO] [1545256735.586315444]: /camera/driver - Starting camera
[ INFO] [1545256735.592578755]: /camera/driver - Publishing camera transforms (/tf_static)
[ INFO] [1545256735.592727578]: /camera/driver - Setting dynamic camera options
**[ERROR] [1545256742.069283310]: /camera/driver - Error calling rs_wait_for_frames ( device:0x109b908 ): 
Timeout waiting for frames.** 

**[camera/nodelet_manager-1] process has died [pid 4222, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/fetch/.ros/log/48698a5a-03d0-11e9-9c86-d8cb8a5dca62/camera-nodelet_manager-1.log].
log file: /home/fetch/.ros/log/48698a5a-03d0-11e9-9c86-d8cb8a5dca62/camera-nodelet_manager-1*.log**
[camera/driver-2] process has finished cleanly
log file: /home/fetch/.ros/log/48698a5a-03d0-11e9-9c86-d8cb8a5dca62/camera-driver-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
 

Don't know how to fix it. Can anyone helps me with this? Thank you so much!

RealSense-Customer-Engineering, 2018-12-25:
[Realsense Customer Engineering Team Comment]
Hi @pl95,

Please use below ROS wrapper and SR300 latest firmware 3.21.0.0.
https://github.com/intel-ros/realsense

pl95, 2019-01-02:
Hi @RealSense-Customer-Engineering 
I follow the instruction and have an make fail when I run 
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release

Base path: /home/fetch/catkin_ws
Source space: /home/fetch/catkin_ws/src
Build space: /home/fetch/catkin_ws/build
Devel space: /home/fetch/catkin_ws/devel
Install space: /home/fetch/catkin_ws/install
####
#### Running command: "cmake /home/fetch/catkin_ws/src -DCATKIN_ENABLE_TESTING=False -DCATKIN_DEVEL_PREFIX=/home/fetch/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/fetch/catkin_ws/install -G Unix Makefiles" in "/home/fetch/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/fetch/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/fetch/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/fetch/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-2.0.3/realsense2_camera)
-- Using these message generators: gencpp;genlisp;genpy
-- Create Debug Build.
-- realsense2_camera: 2 messages, 0 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/fetch/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/fetch/catkin_ws/build"
####
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] [  0%] Built target roscpp_generate_messages_cpp
Built target dynamic_reconfigure_generate_messages_cpp
[  0%] [  0%] Built target sensor_msgs_generate_messages_cpp
Built target std_msgs_generate_messages_cpp
[  0%] Built target _realsense2_camera_generate_messages_check_deps_Extrinsics
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] [  0%] Built target geometry_msgs_generate_messages_py
Built target sensor_msgs_generate_messages_lisp
[  0%] [ 25%] Built target _realsense2_camera_generate_messages_check_deps_IMUInfo
Built target realsense2_camera_gencfg
[ 25%] Built target nodelet_generate_messages_py
[ 25%] [ 25%] Built target geometry_msgs_generate_messages_lisp
Built target tf_generate_messages_py
[ 25%] Built target sensor_msgs_generate_messages_py
[ 25%] Built target _catkin_empty_exported_target
[ 25%] [ 25%] [ 25%] Built target roscpp_generate_messages_py
Built target bond_generate_messages_py
Built target roscpp_generate_messages_lisp
[ 25%] Built target rosgraph_msgs_generate_messages_py
[ 25%] [ 25%] Built target nodelet_generate_messages_cpp
Built target rosgraph_msgs_generate_messages_lisp
[ 25%] Built target bond_generate_messages_cpp
[ 25%] [ 25%] Built target std_msgs_generate_messages_py
Built target bond_generate_messages_lisp
[ 25%] [ 25%] Built target rosgraph_msgs_generate_messages_cpp
Built target tf_generate_messages_lisp
[ 25%] [ 25%] [ 25%] Built target diagnostic_msgs_generate_messages_lisp
Built target tf_generate_messages_cpp
Built target nodelet_generate_messages_lisp
[ 25%] Built target geometry_msgs_generate_messages_cpp
[ 25%] [ 25%] Built target tf2_msgs_generate_messages_cpp
Built target diagnostic_msgs_generate_messages_cpp
[ 25%] [ 25%] Built target actionlib_msgs_generate_messages_cpp
[... 11 lines cut ...]
[ 56%] Built target realsense2_camera_generate_messages_lisp
[ 68%] Built target realsense2_camera_generate_messages_cpp
[ 68%] Built target realsense2_camera_generate_messages
[ 75%] Building CXX object realsense-2.0.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o
/home/fetch/catkin_ws/src/realsense-2.0.3/realsense2_camera/src/base_realsense_node.cpp:384:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
 #pragma omp parallel for schedule(dynamic)
 ^
In file included from /usr/local/include/librealsense2/hpp/rs_device.hpp:8:0,
                 from /usr/local/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/local/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/local/include/librealsense2/rs.hpp:9,
                 from /home/fetch/catkin_ws/src/realsense-2.0.3/realsense2_camera/src/../include/../include/realsense_node_factory.h:11,
                 from /home/fetch/catkin_ws/src/realsense-2.0.3/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/fetch/catkin_ws/src/realsense-2.0.3/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/local/include/librealsense2/hpp/rs_sensor.hpp: In instantiation of ‘void rs2::frame_callback<T>::on_frame(rs2_frame*) [with T = rs2::asynchronous_syncer; rs2_frame = rs2_frame]’:
/home/fetch/catkin_ws/src/realsense-2.0.3/realsense2_camera/src/base_realsense_node.cpp:1394:1:   required from here
/usr/local/include/librealsense2/hpp/rs_sensor.hpp:109:13: error: no match for call to ‘(rs2::asynchronous_syncer) (rs2::frame)’
             on_frame_function(frame{ fref });
             ^
make[2]: *** [realsense-2.0.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[1]: *** [realsense-2.0.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I am using the 2.0.3 release. Also, how can I get the firmware version and how to update to the latest version?

RealSense-Customer-Engineering, 2019-01-03:
[Realsense Customer Engineering Team Comment]
Hi @pl95,

Which librealsense version did you install? Please try to build and install LibRealSense v2.10.2. The latest librealsense version is not compatible with ros wrapper 2.0.3.

You can install SR300 DCM and it will upgrade your firmware automatically.
https://downloadcenter.intel.com/download/25044/Intel-RealSense-Depth-Camera-Manager?product=92329

pl95, 2019-01-07:
Hi @RealSense-Customer-Engineering ,
Thank you so much for your help and patience! The problem is fixed  with the new librealsense installed. 
However, when When I tried to run
roslaunch realsense2_camera rs_rgbd.launch 
I have the following error:

$ roslaunch realsense2_camera rs_rgbd.launch
... logging to /home/fetch/.ros/log/f747509c-129a-11e9-b6f3-d8cb8a5dca62/roslaunch-fetch6-21906.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Unknown substitution command [eval depth_registered_processing and sw_registered_processing]. Valid commands are ['find', 'env', 'optenv', 'anon', 'arg']
The traceback for the exception was written to the log file

How can I fix this error?

RealSenseCustomerSupport, 2019-01-23:
Hi @pl95,

How about below command? can it run normally?
$ roslaunch realsense2_camera rs_camera.launch

pl95, 2019-01-23:
Hi,

 $ roslaunch realsense2_camera rs_camera.launch
This command works well.

According to the website, http://wiki.ros.org/roslaunch/XML, eval is an new
expression in Kinetic. And I am using ROS indigo. So I changed line 161 to
189 in rs_rgbd.launch as following.

 <group if="$(arg depth_registered_processing)">
      <group if="$(arg sw_registered_processing)">
      <node pkg="nodelet" type="nodelet" name="register_depth"
            args="load depth_image_proc/register $(arg manager) $(arg
bond)" respawn="$(arg respawn)">
        <remap from="rgb/camera_info"             to="$(arg
rgb)/camera_info" />
        <remap from="depth/camera_info"           to="$(arg
depth)/camera_info" />
        <remap from="depth/image_rect"            to="$(arg
depth)/image_rect_raw" />
        <remap from="depth_registered/image_rect" to="$(arg
depth_registered)/sw_registered/image_rect_raw" />
      </node>

      
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_sw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager)
$(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg
rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg
rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg
depth_registered_filtered)/sw_registered/image_rect_raw" />
        <remap from="depth_registered/points"     to="$(arg
depth_registered)/points" />
      </node>
      </group>
    </group>

    <group if="$(arg depth_registered_processing)">
      <group if ="$(arg hw_registered_processing)">
      
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_hw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager)
$(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg
rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg
rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg
depth_registered)/image_raw" />
        <remap from="depth_registered/points"     to="$(arg
depth_registered_pub)/points" />
      </node>
      </group>
    </group>

The modified file runs without error. However, it seems that there is no
point cloud message. Following are the camera topics after I run
rs_rgbd.launch.

/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_color/image_raw/theora
/camera/aligned_depth_to_color/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/theora/parameter_updates
/camera/aligned_depth_to_infra1/camera_info
/camera/aligned_depth_to_infra1/image_raw
/camera/aligned_depth_to_infra1/image_raw/compressed
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_infra1/image_raw/compressedDepth
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_infra1/image_raw/theora
/camera/aligned_depth_to_infra1/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/theora/parameter_updates
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color_rectify_color/parameter_descriptions
/camera/color_rectify_color/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra1/image_rect_raw/theora
/camera/infra1/image_rect_raw/theora/parameter_descriptions
/camera/infra1/image_rect_raw/theora/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/realsense2_camera_manager/parameter_descriptions
/camera/realsense2_camera_manager/parameter_updates

Is the rostopics looks right? Which of this topic should be the point cloud
message? Or did I modified the code incorrectly?

Thank you so much for your patience and help!

RealSenseCustomerSupport, 2019-01-24:
Hi@pl95,

Try below command to enable pointcloud and see how it works. By the way, what's the librealsense version you currently use?
$ roslaunch realsense2_camera demo_pointcloud.launch

pl95, 2019-01-24:
Hi,

demo_pointcloud.launch runs without error. But there is no image. And I am
using librealsense2.10.2.
Anything I can do to check what goes wrong?

RealSenseCustomerSupport, 2019-01-25:
Hello @pl95,

1. Please also show me your log when you run the command. you don't see the visualized 3D map showing in rviz?
2. The realsense-viewer can run the depth and RGB streams normally, right?
3. Did you upgrade the SR300 firmware to 3.21.0.0?

pl95, 2019-01-28:
Hi @RealSenseCustomerSupport ,

**1. $lsusb -t**
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/2p, 5000M
        |__ Port 1: Dev 4, If 0, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 1: Dev 4, If 1, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 1: Dev 4, If 2, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 1: Dev 4, If 3, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 1: Dev 4, If 4, Class=Vendor Specific Class, Driver=, 5000M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/2p, 480M
        |__ Port 2: Dev 4, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 2: Dev 4, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 2: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 5, If 0, Class=Audio, Driver=snd-usb-audio, 12M
    |__ Port 3: Dev 5, If 1, Class=Audio, Driver=snd-usb-audio, 12M
    |__ Port 3: Dev 5, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 7: Dev 6, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 7: Dev 6, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=uhci_hcd/2p, 12M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=uhci_hcd/2p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/4p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M

**$lsusb -v**
Bus 002 Device 002: ID 8087:8001 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x8087 Intel Corp.
  idProduct          0x8001 
  bcdDevice            0.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
[... 1994 lines cut ...]
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval             255

**2.** I don't have htop installed. So I run **"top"** instead
[image]

Following is when demo_pointcloud.launch is not running.
[image]

3. I think these command are used in Windows system. My computer has Ubuntu 14.04 system. If I run **$intel_rs_dcm_sr300_3.3.27.5718.exe --extract-to=\home\fetch\ --extract-only**
intel_rs_dcm_sr300_3.3.27.5718.exe: command not found

4.The warning message "Empty frame has arrived" didn't happen every time. And it usually stop printing that warning message after few second.

5. Since my computer is using Ubuntu 14.04 system with ROS indigo, and the newest wrapper don't support ROS indigo, I used the newest wrapper that can support indigo. Should I tried using the newest wrapper anyway?

RealSenseCustomerSupport, 2019-01-31:
----------------------------------------------

Freeman Lo, Jan 31, 2:39 PM

Hi @pl95,

If you have other items' update, let me know.

RealSenseCustomerSupport, 2019-02-27:
----------------------------------------------

Hi @pl95,

Did you try the latest ROS wrapper on ROS Kinetic? If no additional support required, will close this.

pl95, 2019-02-27:
Hi,

I didn't try the ROS wrapper on ROS Kinetic. Because I have ROS indigo
instead of ROS Kinetic, and everything else on the computer is build for
Indigo.  I don't think I can upgrade the ROS version to Kinetic. I do need
additional support. When I run the rgbd launch file, it sometimes gives me
the following error.

... logging to
/home/fetch/.ros/log/09768c8e-3aa4-11e9-aa0b-d8cb8a5dca62/roslaunch-fetch6-27004.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fetch6:40598/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/camera_aligned_depth_to_color_frame_id:
camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id:
camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra1_frame_id:
camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra2_frame_id:
camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path:
 * /camera/realsense2_camera/serial_no:
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
[... 191 lines cut ...]

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)
    register_depth (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera/realsense2_camera_manager-1]: started with pid [4728]
process[camera/realsense2_camera-2]: started with pid [4729]
process[camera/color_rectify_color-3]: started with pid [4730]
process[camera/register_depth-4]: started with pid [4740]
process[camera/points_xyzrgb_sw_registered-5]: started with pid [4749]
process[camera/points_xyzrgb_hw_registered-6]: started with pid [4751]
[ INFO] [1551299271.647355610]: Initializing nodelet with 4 worker threads.

I will need the pointcloud data later. Could you help me figure out what is
wrong with this?

Thank you so much!

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi @pl95,

1. From "top" log, it looks the overall cpu consumption/loading is quite heavy when you run "demo_pointcloud".  Please run also "lscpu" to show the info.
2. Try below command to see how it works.
$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
3. Indeed, above DCM exe command is used for Windows to upgrade your SR300 firmware.
4. So far, I don't have Ubuntu 14.04 + Indigo environment. Will try to setup to see how it goes.

pl95, 2019-03-07:
HI,

1. The "lscpu" log when "demo_pointclould" is *running:*
*$ lscpu*
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    1
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 60
Stepping:              3
CPU MHz:               3226.816
BogoMIPS:              5799.92
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              6144K
NUMA node0 CPU(s):     0-3

"lscpu" log when "demo_pointcloud" *NOT running:*
*$ lscpu*
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    1
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 60
Stepping:              3
CPU MHz:               3306.453
BogoMIPS:              5799.92
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              6144K
NUMA node0 CPU(s):     0-3

*2. $ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud*
... logging to
/home/fetch/.ros/log/c5bee004-410b-11e9-b06f-d8cb8a5dca62/roslaunch-fetch6-16790.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fetch6:40161/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/camera_aligned_depth_to_color_frame_id:
camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id:
camera_aligned_de...
[... 83 lines cut ...]
[ INFO] [1551988396.169954137]: infra1 stream is enabled - width: 640,
height: 480, fps: 30
[ INFO] [1551988396.170957074]: color stream is enabled - width: 640,
height: 480, fps: 30
 07/03 14:53:16,177 WARNING [140205670668032] (backend-v4l2.cpp:808) Empty
frame has arrived.
 07/03 14:53:16,177 WARNING [140205670668032] (backend-v4l2.cpp:808) Empty
frame has arrived.
 07/03 14:53:16,177 WARNING [140205670668032] (backend-v4l2.cpp:808) Empty
frame has arrived.
 07/03 14:53:16,177 WARNING [140205670668032] (backend-v4l2.cpp:808) Empty
frame has arrived.
[ INFO] [1551988396.178935074]: publishStaticTransforms...
[ INFO] [1551988396.179070655]: RealSense Node Is Up!
 07/03 14:53:16,182 WARNING [140205659801344] (backend-v4l2.cpp:808) Empty
frame has arrived.
 07/03 14:53:16,182 WARNING [140205659801344] (backend-v4l2.cpp:808) Empty
frame has arrived.
 07/03 14:53:16,182 WARNING [140205659801344] (backend-v4l2.cpp:808) Empty
frame has arrived.
 07/03 14:53:16,182 WARNING [140205659801344] (backend-v4l2.cpp:808) Empty
frame has arrived.

Raw image, depth image and infra image are received. But poincloud data is
not received.

pl95, 2019-03-08:
Hi,

I re-installed the point cloud package and check the image when I run the
demo_point cloud.  It shows images now when I run the demo. But I still not
have any image data for rs_rgbd.launch,

Is this information helps you solve the problem? Thank you so much for your
patience!

freemanlo, 2019-03-09:
Hi @pl95 
Is demo_pointcloud.launch can't fit your usage? Instead of rs_rgbd.launch, you can also try below to have depth/ir/color images and visualize pointcloud via rviz.
$ roslaunch realsense2_camera rs_camera.launch enable_pointcloud:=true

pl95, 2019-03-25:
HI @RealSenseCustomerSupport ,

I change the launch file rs_rgbd.launch as following, which match the demo_pointcloud.launch. 

<launch>
  <arg name="namespace"           default="camera"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  

  <arg name="serial_no"           default=""/>
  <arg name="json_file_path"      default=""/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  **<arg name="enable_fisheye"      default="false"/>**

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  **<arg name="enable_depth"        default="true"/>**

  <arg name="infra1_width"        default="640"/>
  <arg name="infra1_height"       default="480"/>
  **<arg name="enable_infra1"       default="false"/>**

  <arg name="infra2_width"        default="640"/>
  <arg name="infra2_height"       default="480"/>
  **<arg name="enable_infra2"       default="false"/>**

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra1_fps"          default="30"/>
  <arg name="infra2_fps"          default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="1000"/>
  <arg name="accel_fps"           default="1000"/>
  **<arg name="enable_imu"          default="false"/>**

  **<arg name="enable_pointcloud"   default="true"/>
  <arg name="enable_sync"         default="true"/>**
  <arg name="align_depth"         default="true"/>

  

  
  <arg name="rgb"                             default="color" />
  <arg name="ir"                              default="infra1" />
  <arg name="depth"                           default="depth" />
  <arg name="depth_registered_pub"            default="depth_registered" />
  <arg name="depth_registered"                default="depth_registered" unless="$(arg align_depth)" />
  <arg name="depth_registered"                default="aligned_depth_to_color" if="$(arg align_depth)" />
  <arg name="depth_registered_filtered"       default="$(arg depth_registered)" />
  <arg name="projector"                       default="projector" />

  
  <arg name="bond"                            default="false" />
  <arg name="respawn"                         default="$(arg bond)" />

  
  <arg name="rgb_processing"                  default="true"/>
  <arg name="debayer_processing"              default="false" />
  <arg name="ir_processing"                   default="false"/>
  <arg name="depth_processing"                default="false"/>
  <arg name="depth_registered_processing"     default="true"/>
  <arg name="disparity_processing"            default="false"/>
  <arg name="disparity_registered_processing" default="false"/>
[... 62 lines cut ...]
      </node>

      
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_sw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered_filtered)/sw_registered/image_rect_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered)/points" />
      </node>

      
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_hw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered)/image_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered_pub)/points" />
      </node>

  </group>

</launch>

Now, I have the image on with the robot model. I don't know whether the problem should be solved in this way. But I think it is something I can try.

closed 2019-03-25 by pl95