realsenseai/librealsense #2941: PLY-files without directly using the viewer?

Schnitzelmon, 2018-12-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Win 10  | 

| Platform                                 | PC  |
| SDK Version                            | 2.0                | 
| Language                            |  C/C#/labview/nodejs/opencv/python                       | 
| Segment			|  others                          | 

### Issue Description
Hello, I am using the Intel RealSense D435 Camera to generate ply pointclouds. I am doing this on a PC with the Intel RealSense Viewer with the "export 3D model to PLY format"-button (I am using the Post-Processing there). 

So my question is: Is there a way to make this easier and maybe automatic? I mean like a command for batch file or a code to do this without directly using the viewer.
I looked through the SDK files and haven't found anything for just saving the PLY-file. 

TLDR: I am looking for an easy way to generate the PLY files (maybe with Post-Processing)

Thanks and greetings 
Felix

dorodnic, 2018-12-20:
Hi @Schnitzelmon 
You can add to [rs-pointcloud](https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp#L49):
```
points.export_to_ply("filename.ply", color);
```
Same can be done in python and other languages.

Schnitzelmon, 2018-12-21:
Thanks a lot for your help. It worked fine. I used python. Awesome

closed 2018-12-24 by dorodnic

duongbaoquy, 2020-03-13:
I have a question about this topic
if we use the command "points.export_to_ply (" {file_name} .ply, color "
then it will export to a ply file but in the form of 'mesh', also known as Polygon. So my question is, do we have a way to export a pointcloud file (we can see the points)?
By the way, i'm using realsense camera D435i