realsenseai/librealsense #2942: Invalid frame size

eprat04, 2018-12-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|-------------------------------- |------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |  Linux (Ubuntu)  | 
| Kernel Version (Linux Only)    |  3.10.40 | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  2.17.0  | 
| Language                            |  C++ | 

### Issue Description
I am  doing following test (using bulk mode transfer):

while(1) {
pipe.start();
pipe.wait_for_frames();
pipe.stop;
sleep(2s);
}

and, sometimes, received frame size is equal to 65536 pixels, instead of 1843200 (for 1280 * 720 resolution) or 614400 (for 640 * 480 resolution).

This issue is reproduced when changing stream configuration (resolution, format, ...).

It could lead to segmentation fault in librealsense::unpack_yuy2 since frame size is not checked before unpacking the received frame :

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xa8e49450 (LWP 3089)]
0xb6ba7b7e in librealsense::unpack_yuy2<(rs2_format)5> (d=0xb1d00480, 
    s=0xb1d01470 "\\\177]\206]\177^\206]~]\206]~_\206^\177]\207^\200^\210^\200^\207_\201_\207_\201_\207_\203`\207`\202`\207a\201a\207b\201a\207a\201b\207c\177b\207b\200b\207b\200c\207c\177d\206c\177c\205c\200c\206c\201d\207c\201b\207a\201c\207d\201c\206c\201c\206c\201c\206d\200c\206c\200d\207d\200e\206c\201d\206d\201d\206c\177c\205d\200d\205d\177e\206e\177d\207c\200c\210d\177d\210d\177d\210d~c\210b\177b\207c\200c\206b\200c\206c\200b\206b\177b\207a\177c\210b\177b\210c\200b\211a\200a\207a\200a\205a\200`\205"..., width=640, height=480)
    at /home/ubuntu/uvc/realsense/librealsense-2.17.0/src/image.cpp:517
517	            }, u[16] = {
(gdb) bt
#0  0xb6ba7b7e in librealsense::unpack_yuy2<(rs2_format)5> (d=0xb1d00480, 
    s=0xb1d01470 "\\\177]\206]\177^\206]~]\206]~_\206^\177]\207^\200^\210^\200^\207_\201_\207_\201_\207_\203`\207`\202`\207a\201a\207b\201a\207a\201b\207c\177b\207b\200b\207b\200c\207c\177d\206c\177c\205c\200c\206c\201d\207c\201b\207a\201c\207d\201c\206c\201c\206c\201c\206d\200c\206c\200d\207d\200e\206c\201d\206d\201d\206c\177c\205d\200d\205d\177e\206e\177d\207c\200c\210d\177d\210d\177d\210d~c\210b\177b\207c\200c\206b\200c\206c\200b\206b\177b\207a\177c\210b\177b\210c\200b\211a\200a\207a\200a\205a\200`\205"..., width=640, height=480)
    at /home/ubuntu/uvc/realsense/librealsense-2.17.0/src/image.cpp:517
#1  0xb6b58ffe in librealsense::uvc_sensor::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)>::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>) (__closure=0xb1d005b0, p=..., 
    f=..., continuation=...) at /home/ubuntu/uvc/realsense/librealsense-2.17.0/src/sensor.cpp:510
...

Has anyone observed such kind of issue?
Do you know how this issue could be fixed?

alex-precosky, 2018-12-20:
Hi eprat04, this might be the same issue I've been getting. Does it happen most with higher image resolutions? I get it about every second frame when using 1280x720 or 1920x1080 resolutions: https://github.com/IntelRealSense/librealsense/issues/2682

I've been getting support on nvidia's NVOnline as well, but haven't been able to resolve the problem. Best I can do is detect when it happens, and discard these frames so that they don't affect downstream processes. At least for me, the problem seems upstream of librealsense - it happens with plain V4L2 applications (which librealsense uses as a backend on linux),

dorodnic, 2018-12-20:
Hi @eprat04 @alex-precosky 
Are you using the default V4L2 backend + kernel patch for 3.10, or are you building the SDK from source with `-DFORCE_LIBUVC=true`?

alex-precosky, 2018-12-20:
I haven't tried running the cameras with a librealsense compiled -DFORCE_LIBUVC=true, but I will try that and see if that helps. Thanks for the idea!

eprat04, 2019-01-02:
Hi @dorodnic @alex-precosky
Thanks a lot for your comments.
Yes, it seems that this issue happens more often with higher resolutions.
And I am building the SDK with FORCE_LIBUVC option.

RealSense-Customer-Engineering, 2019-01-02:
[Realsense Customer Engineering Team Comment]
Hi @eprat04,

Did you use the Jetson Tx2 platform? If so, you can refer to below and have the required kernel patch instead of using -DFORCE_LIBUVC=true. This should be more stable.
https://github.com/jetsonhacks/buildLibrealsense2TX

eprat04, 2019-01-03:
Hi @RealSense-Customer-Engineering 

Unfortunately, I have to build the SDK using -DFORCE_LIBUVC=true option.
I can avoid the segmentation fault by dropping the incomplete frames (as done in kernel patch https://github.com/jetsonhacks/buildLibrealsense2TX/blob/master/patches/incomplete-frame.patch),
but I just wonder if the root cause of this issue is known and if there is a fix available.

RealSenseCustomerSupport, 2019-01-22:
Hi @eprat04,

Do you use JetPack 3.3 + L4T 28.2.1? And, you upgrade to Ubuntu 18.04?
If you initialize the color resolution to VGA and YUYV format, will the incomplete frames still happen?
If it happens, please also have the librealsense log ($ export LRS_LOG_LEVEL=DEBUG) and dmesg log.

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

Hi @eprat04,

Please consider to upgrade your BSP to L4T 28.2.1 and apply the patches. Note: On the NVIDIA Jetson TX1/2, high frame rates can return incomplete frames.
https://github.com/jetsonhacks/buildLibrealsense2TX/tree/master/patches

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi @eprat04,

Any good progress to resolve this?

eprat04, 2019-03-07:
Hi @RealSenseCustomerSupport,

As explained in comment posted on January 3rd, I already implemented a workaround to avoid any segmentation fault.
Do you know why this issue occurs on Jetson TX1/2?

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

Hi @eprat04,

For the workaround "incomplete-frame.patch", two comments:
1. From the usb log, it look like the whole sw stack process take longer time than expected. Ideally, 30ms per frame (based on 30FPS), but it takes longer process time (ex: 66ms). It causes the incomplete video frame happens frequently, especially higher resolution like 720p at 30FPS for color stream.
2. Enabling CUDA build option should reduce the conversion/processing time of video frames

eprat04, 2019-03-11:
Hi @RealSenseCustomerSupport,

Thanks a lot for this information.
I will try to do the test with CUDA enabled, I'll keep you informed.

RealSenseCustomerSupport, 2019-04-12:
----------------------------------------------

Hi @eprat04,

Let me know if any update. If no further question, will close this.

eprat04, 2019-04-12:
Hi @RealSenseCustomerSupport,

No update at this time.

closed 2019-05-10 by RealSenseCustomerSupport