realsenseai/librealsense #2944: Poor Depth Output with D435. Defective Device?

kylehounslow, 2018-12-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.6.0 | 
| Operating System & Version | (Ubuntu 16.04 LTS) | 
| Kernel Version (Linux Only)    | 4.15.0-42-generic                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.17.0                      | 
| Language                            |  C/C++/Python                           | 
| Segment			|  VR/AR                         | 

### Issue Description
I have purchased two D435 cameras and one has significantly better depth output. I have followed the whitepapers for tuning the parameters still not even close to getting comparable results.  
Is my device defective? Anyone seeing same issues?

Below is `realsense-viewer` output from each D435 with Default settings:

D435 with poor depth
[image]

D435 good depth
[image]

Any help is appreciated, thanks.

dorodnic, 2018-12-21:
Hi @kylehounslow 
You wrote 5.10.6.0 under firmware, could you please confirm this is correct for both devices?

kylehounslow, 2018-12-21:
Update: I tried flashing very latest firmware 5.10.6.13 to both devices and still the same result.

dorodnic, 2018-12-22:
IMHO, there might be something wrong with the infrared projector. Such defect should not have passed the production line tests, but perhaps the projector or the lens got damaged after manufacture. 

Did you try adjusting laser power? What are the compared IR images you get from the two devices? 

Perhaps @RealSense-Customer-Engineering could also comment, as they have more experience with the hardware aspect.

Regardless, I believe you have the option to return / replace the device (exact details depend on the re-seller)

RealSense-Customer-Engineering, 2018-12-27:
[Realsense Customer Engineering Team Comment]
Hi @kylehounslow,

Have you tried to re-calibrate this unit by OEM/Dynamic Calibration tool or called`Intel.Realsense.CustomRW -g` to restore it back to golden factory standard?
https://downloadcenter.intel.com/download/27955/Intel-RealSense-D400-Series-Calibration-Tools-and-API?v=t

kylehounslow, 2018-12-31:
@RealSense-Customer-Engineering @dorodnic 
Great news: recalibration of the device using `Intel.Realsense.DynamicCalibrator` worked and the depth quality is virtually the same for both cameras.
  
D435 with previously poor depth from above, After recalibration:
[image]  
  
Compared with D435 with previously good depth from above:
[image]

Is there known causes for the D435 cameras to become "un-calibrated"? I'd like to avoid having to do this again if possible.

RealSense-Customer-Engineering, 2019-01-16:
[Realsense Customer Engineering Team Comment]
I am glad that the recalibration works.
All the units were pre-calibrated in the factory before they were shipped out. However, depending on the local shipping and handling method and other various reasons, sometimes such as this case, re-calibration is needed.
I will close this ticket for now.
Thank you.

closed 2019-01-16 by RealSense-Customer-Engineering