realsenseai/librealsense #2945: How to get depth information and IMU data at the same time on windows ?

MaoPann, 2018-12-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |            | 
| Platform                                 | PC.  |
| SDK Version                            | sdk 2.6.5                         | 
| Language                            |  {C }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
<>

MaoPann, 2018-12-22:
In the enable_stream function, how should I configure it?

MaoPann, 2018-12-22:
Can you share a call code?

MaoPann, 2018-12-22:
Thank you very much for your guidance.

dorodnic, 2018-12-22:
Happy to help. 

To enable IMU explicitly, you can call `config.enable_stream(RS2_STREAM_ACCEL);` and `config.enable_stream(RS2_STREAM_GYRO);`.

You could also specify the frame-rate using `config.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F, 250)` (this will set the accelerometer rate at 250 Hz)

You can see the list of supported configurations [here](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-motion.h#L49-L53) and in the device data-sheet. 

We don't include explicit configuration details in the examples, because we want the same examples to work on all supported (and future) devices. So if you don't provide an explicit configuration, you will just get the default IMU from the D435i.

Once you do get a frameset with motion frames, you can cast them to [rs2::motion_frame](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_frame.hpp#L772-L798):
```cpp
using namespace std;
using namespace rs2;
frameset fs = p.wait_for_frames();
for (auto& f : fs) {
   if (f.is<motion_frame>()) {
       motion_frame mf = f.as<motion_frame>();
       rs2_vector xyz = mf.get_motion_data();
       cout << f.get_profile().stream_type() << ": " 
            << xyz.x << ", " << xyz.y << ", " << xyz.z << endl;
   }
}
```