realsenseai/librealsense #2946: Building Examples only

abdelrahmanhussein, 2018-12-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D435 | 
| Firmware Version                   | 5.10.13 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-42-generic |                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.17.0                         | 
| Language                            |  C                          | 
| Segment			|  others                           | 

### Issue Description
Hi,

**I have successfully installed the realsense-viewer. I have also successfully build the libsense-2 and was able to run the examples. Everything works fine!**

My question is regarding building examples independently without having to building the whole library again. Assume that I modified the source code of an example to do different functionality, do I have to go through the whole flow of CMaking and building the whole library again? Or there is a way in which I can just compile the example source code after modifying it?

After installing everything, I tried to run CMake for rs-depth example by carrying out the following steps:

1. go to the directory "/librealsense-master/examples/C/depth"
2. Create a new directory called "build" and then `cd build`
3. Run the command `cmake ../ -DBUILD_EXAMPLES=true`

doing this, I got the following output

> -- The C compiler identification is GNU 5.5.0
> -- The CXX compiler identification is GNU 5.5.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> CMake Error at CMakeLists.txt:15 (install):
>   install TARGETS given no RUNTIME DESTINATION for executable target
>   "rs-depth".
> 
> 
> -- Configuring incomplete, errors occurred!
> 

So, what was wrong about what I did? And how to build the example alone?

closed 2018-12-21 by abdelrahmanhussein

jacksonkr, 2019-01-29:
Do I edit the CMake files directly in order to chose the examples I want built? Also, what if the examples I need require something like the aforementioned `rs-depth`?

milan-r-shah, 2020-07-09:
> Hi @abdelrahmanhussein
> After the initial `cmake ..` and `make`, you can modify the examples and just run `make` again in the build folder. This will only build what changed, and not rebuild the SDK.

So, to confirm, by "just run `make` again in the build folder", you meant run `make` in `librealsense/build` directory?