realsenseai/librealsense #2947: No video frame returns in Android system

David-168, 2018-12-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | D435
| Firmware Version                   | V5.10.13) | V5.10.13
| Operating System & Version |  Android 8.0.0 | Android 8.0.0
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Qcom845/Zenphone |Qcom845/Zenphone ASUS_Z01RD
| SDK Version                            | 2.16.5                        | 2.16.5
| Language                            |   C                   | C
| Segment			|  Smartphone/AR                           | Smart phone/AR

### Issue Description
<Describe your issue / question / feature request / etc..>
1. No RGB/YUV frame returns on current libuvc configuration. ( The previous setting works fine on my platform. ) . It fails at profile = pipe.start(config); (native-lib.cpp)

            static const rs2_format format = RS2_FORMAT_YUYV;
            static const rs2_stream stream = RS2_STREAM_COLOR;
........
                    rs2::frameset data = pipe.wait_for_frames();
//                    auto rgb_vid_frame = colorizer.colorize(data.get_depth_frame());

                    env->SetByteArrayRegion(arr, 0, rgb_frame_size,
                                            reinterpret_cast<const jbyte *>(data.get_data()));

2.  No device found on V4L2 configuration. ( This is suggestion from previous issues. ) It fails at 
        auto list = ctx.query_devices();  (native-lib.cpp) No device found. 

Here is modification List.
A.      set(FORCE_LIBUVC OFF)    in   librealsense\CMake\android_config.cmake
B.     #include <array> in backend-v4l2.h 
C.      chmod 777 * in /sys/class/video4linux
D.  I am trying to check if it works fine for existing video buffer format. So, I didn't modify Z16 buffer part.

3. I am trying to put my APK on Red Hydrogen One ( Qcom835. I need that fancy light field film. ) Could you pre-release the non-root Android code to me? I can help to test it. 

4. The software setting shows that SDK can support RGB frame. But, the document shows the device can only support YUY2. It's written in Intel-RealSense-D400-Series-Datasheet.pdf. 
And, one of example is using RGB format. Is there any color space transformation there ? In software side or in firmware side ?  

thx

RealSense-Customer-Engineering, 2019-01-08:
[Realsense Customer Engineering Team Comment]
@David-168 Did you have any update about the issue? Could you please try as Dorodnic suggested that reduce the RGB resolution? Alternatively, you can add `rs_log_to_file` to your app and share librealsense logs, to understand why RGB frames fail to arrive.

David-168, 2019-01-09:
I tried RS2_FORMAT_RGB8 and Y16 with 424X240. Because the camera supports both Y16 and YUY2 for it. It might save time without up-scaling thing. But, the result is the same. 
The good new is I tried pipe.start() without config setting. And I sync the depth and color buffers together. It works fine.  Maybe it's not the right way to solve this issue. 
I tried to use same setting( 640X480/Y16/RGB8 and call  pipe.start(config). The result is not good.
Since default configuration with frame sync works fine. I am going to stick with it for a while. 
As you mention earlier, color space conversion is done by software. I will try to move this part to my OpenGLES code.

David-168, 2019-01-09:
This issue has been solved.

closed 2019-01-09 by David-168

RealSense-Customer-Engineering, 2019-01-14:
[Realsense Customer Engineering Team Comment]
@David-168 Any update about the issue? Looking forward to your reply. Thanks!

RealSense-Customer-Engineering, 2019-01-16:
[Realsense Customer Engineering Team Comment]
@David-168 Sorry that we're having some problem to sync from github to our internal support platform. Now I noticed you closed it. Thanks!