realsenseai/librealsense #2948: relationship between a point in RGB Image and its corresponding in depth Image

Medo91code, 2018-12-21:
Hello 

I use D435 and I should detect a shape (a point "Pixel" in this shape ) on RGB Image with resolution 1920X1080 . we will say the coordinate of this point ( the coordinate of red Point in RGB image) is 1000 X 700 .
 after that I will get a distance with:
rs2::depth_frame dpt_frame = frame.as<rs2::depth_frame>();
float pixel_distance_in_meters = dpt_frame.get_distance(x,y);

but the pixel 740 X 500 in depth image represent this Point not the pixel 1000 X 700. 

There is displacement and other Transformations between RGB Image and depth Image .

I want to know the Mathematical formulas or Codes / Methods to Extract these displacement and Transformations .

I tested 3 examples 
1- rs-align.cpp
2- rs-measure.cpp
3-rs-pointcloud.cpp

these 3 examples are great but with these i have "aligned Image" or " Point cloud Image"
 with high noise  (i want firstly to detect the object only from RGB Image after that i use depth Imege / get_distance(x,y); ).

Important note :  I noticed that the depth image represent the same points as will as the left_IR Image .
i mean   " no displacement and other Transformations between  left_IR Image and depth Image " .
is this note  correct ?
if it is correct , Why ?

Attached files :  
1- Rgb Image
2- depth
3_ Left ir imager

[image]
[image]
[image]

dorodnic, 2018-12-21:
Hi @Medo91code 

The mathematics of these transformations are concentrated in [rsutil.h](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h).
There is decent documentation on this topic [in our wiki](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0)

Specifically - for the D400 camera, depth image is generated from the left & right infrared and is perfectly aligned to the left infrared image.

Given pixel and its depth, you can [convert it to 3D point](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L49-L69):
First, move to lens coordinates by subtracting the principal point and dividing by focal length.
Next, apply any lens specific distortion (for D400 there is none)
Finally, the 3D point will be (x * d, y * d, d)

You can query extrinsics between any two streams - this will give you are rotation and translation between the two sensors. [Multiply by the resulting matrix](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L71-L77) and [do the whole procedure in reverse](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L17-L46), and you get a pixel in the color image.

Let me know if this answer your question

Medo91code, 2018-12-22:
Thank you for your response .

i know this take some time  from you , but please can you add an practical article about (rs2_project_color_pixel_to_depth_pixel ) in here( https://github.com/IntelRealSense/librealsense/wiki).
i find this point very important for me and for all realsense_programmer.
or please can you provide me a practical  Example or tell me what are these parameters in the function (rs2_project_color_pixel_to_depth_pixel) and from where can i pass their values :  

the parameter are :  

float to_pixel[2],
const uint16_t* data, float depth_scale,
float depth_min, float depth_max,
const struct rs2_intrinsics* depth_intrin,
const struct rs2_intrinsics* color_intrin,
const struct rs2_extrinsics* color_to_depth,
const struct rs2_extrinsics* depth_to_color,
const float from_pixel[2])

thank you

yuemasy, 2018-12-25:
@Medo91code https://github.com/IntelRealSense/librealsense/blob/79dda7f00ebde9564f1a2053b8ab95b7ba41ac69/unit-tests/unit-tests-live.cpp#L4936 ， 
if you ask this

dorodnic, 2018-12-25:
Hi @Medo91code 
To get extrinsics / intrinsics, please see [wiki/API-How-To](https://github.com/IntelRealSense/librealsense/wiki/API-How-To).
Depth min can be 0, depth max depends on your application (in meters). 10 should be plenty for now. 
Depth scale can be queried from the device, but the default for D400 is 0.001. 
from_pixel should be the x and y of the pixel you want to transform. 
to_pixel will store the result

Medo91code, 2018-12-30:
Hi 
 I used  (rs2_project_color_pixel_to_depth_pixel) and the code works fine .

Thank you @dorodnic  @yuemasy  @RealSense-Customer-Engineering

closed 2019-01-14 by RealSense-Customer-Engineering

95ImadEL, 2019-10-17:
@dorodnic please should I perform depth_rgb alignment before using the function rs2_project_color_pixel_to_depth_pixel() ? Thank you in advance.

VD2410, 2020-01-22:
Hello, @Medo91code @dorodnic.
I have a similar problem, and I want to know how did you use (rs2_project_color_pixel_to_depth_pixel)?
Thank you 
 Vishal

kumarvis, 2020-02-06:
Hello @Medo91code  @dorodnic  Hey I have problem in revere manner I want to project (Depth Pixel -> Color Pixel) in way reverse of rs2_project_color_pixel_to_depth_pixel. Please help. I did the reverse of 
Depth Pixel - > Depth point
Depth Point  - > Color Point
Color Point -> Color Pixel 
But the X alignment has significant displacement.