realsenseai/librealsense #2955: openni::CoordinateConverter can't convert coordinates correctly

UnaNancyOwen, 2018-12-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400                                                      | 
| Firmware Version                   | 05.10.13.00                                              | 
| Operating System & Version | Windows 10                                                    | 
| SDK Version                            | 2.17.0                          | 
| Language                            | OpenNI2                          |

### Issue Description
<code>openni::CoordinateConverter</code> can't convert coordinates correctly.
I tried the following APIs. It returns converted coordinates but it is incorrect.

* <code>openni::CoordinateConverter::convertDepthToColor()</code>
* <code>openni::CoordinateConverter::convertDepthToWorld()</code>

Particularly, I think <code>convertDepthToWorld()</code> is important for make point cloud.
@dorodnic @wongfei What do you think? Thanks,

UnaNancyOwen, 2018-12-24:
If it is difficult to fix <code>openni::CoordinateConverter</code>, I think it is necessary to implement conversion processing in rs2driver like https://github.com/IntelRealSense/librealsense/issues/2825#issuecomment-443785509.

wongfei, 2018-12-26:
Needs more investigation, but try this one:

[file Rs2Stream.cpp]
```
OniStatus Rs2Stream::initialize(class Rs2Device* device, rs2_sensor* sensor, int sensorId, int streamId, std::vector<Rs2StreamProfileInfo>* profiles)
{
	rsTraceFunc("type=%d sensorId=%d streamId=%d", (int)m_rsType, sensorId, streamId);

	m_device = device;
	m_sensor = sensor;
	m_sensorId = sensorId;
	m_streamId = streamId;
	m_profiles = *profiles;

	memset(&m_videoMode, 0, sizeof(m_videoMode));
	memset(&m_intrinsics, 0, sizeof(m_intrinsics));

	if (m_oniType == ONI_SENSOR_DEPTH) m_videoMode.pixelFormat = ONI_PIXEL_FORMAT_DEPTH_1_MM;
	else if (m_oniType == ONI_SENSOR_COLOR) m_videoMode.pixelFormat = ONI_PIXEL_FORMAT_RGB888;
	else m_videoMode.pixelFormat = ONI_PIXEL_FORMAT_GRAY8;
	
	m_videoMode.resolutionX = 640;
	m_videoMode.resolutionY = 480;
	m_videoMode.fps = 30;

	// PATCH
	Rs2StreamProfileInfo* spi = getCurrentProfile();
	rs2_get_video_stream_intrinsics(spi->profile, &m_intrinsics, nullptr);
	rs2_fov(&m_intrinsics, &m_fovX);

	if (m_oniType == ONI_SENSOR_DEPTH)
	{
		m_depthScale = rs2_get_option((const rs2_options*)m_sensor, RS2_OPTION_DEPTH_UNITS, nullptr);
	}
	else
	{
		m_depthScale = 0;
	}

	rsLogDebug("type=%d sensorId=%d streamId=%d fovX=%f fovY=%f depthScale=%f",
		(int)m_rsType, m_sensorId, m_streamId, m_fovX, m_fovY, m_depthScale);
	// END PATCH

	return ONI_STATUS_OK;
}

void Rs2Stream::onPipelineStarted()
{
	// EMPTY
}

```

UnaNancyOwen, 2018-12-26:
<code>openni::CoordinateConverter::convertDepthToWorld()</code> seems to returns better output than before.
<code>openni::CoordinateConverter::convertDepthToColor()</code> will still need more investigation. I think that it is not a good result. (especially wide view angle RealSense D435/D435i)

wongfei, 2018-12-27:
[file: Rs2Stream.cpp]
```
OniStatus Rs2Stream::convertDepthToColorCoordinates(StreamBase* colorStream, int depthX, int depthY, OniDepthPixel depthZ, int* pColorX, int* pColorY)
{
...
	rs2_deproject_pixel_to_point(point, &m_intrinsics, pixel, depthZ);
	rs2_project_point_to_pixel(pixel, &((Rs2Stream*)colorStream)->m_intrinsics, point);
...
}
```

another solution to enable depth -> color registration so pixels coords will match:
g_device.setImageRegistrationMode(openni::IMAGE_REGISTRATION_DEPTH_TO_COLOR)

closed 2019-01-15 by RealSense-Customer-Engineering