realsenseai/librealsense #2957: Can't find the librealsense

AdamPengG, 2018-12-25:
I have installed the newest 2.17 sdk. But when I catkin_make the ROS workspace the terminal shows can't find the librealsense.
[image]

dorodnic, 2018-12-25:
Hi @AdamPengG 
Please note that maplab_realsense works with only ZR300, thus requiring librealsense 1.12.1 to be installed. 
If you are using the D400 series perhaps a better approach would be to use `ros_realsense` official wrapper with maplab directly.

AdamPengG, 2018-12-25:
Thanks,Dorodnic. I download this address "https://github.com/intel-ros/realsense"  ROS wrapper.It isn't correct yep,Can you give me a suitable scripts address?

dorodnic, 2018-12-25:
[intel-ros/realsense](https://github.com/intel-ros/realsense) is the official wrapper supporting modern stereo and structured-light RealSense devices. 
What do you mean it isn't correct? 

Could you please clarify what type of camera do you have?

AdamPengG, 2018-12-25:
I have got where is wrong. I have zr300 and D435i.The maplab and ros_realsense in a same workspace.
Can I install the librealsense 1.12.1 and Intel® RealSense™ SDK 2.0 (build 2.17.0) in a same usr?

dorodnic, 2018-12-25:
Got it. There should be no issue installing librealsense 1.12.1 and 2.x side-by-side. 
Regarding D435i, please check [SLAM-with-D435i](https://github.com/intel-ros/realsense/wiki/SLAM-with-D435i). We are still working on an official release

AdamPengG, 2018-12-25:
Thank you Dorodnic. When you release D435i official roslaunch.Please @me in this issue.

closed 2019-01-08 by RealSense-Customer-Engineering