realsenseai/librealsense #2958: Segmentation error in saving a filtered depth image

armandok, 2018-12-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.10.03.00 | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  4.15.0-33                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.17.0                           | 
| Language                            | C++                      | 

I get a segmentation error when I save a filtered and aligned depth image which is converted to `cv::Mat` format.
In particular, the error happens when I use the decimation filter. It doesn't have any problems with the spatial filter though.

Here is my code:

    rs2::pipeline pipe;
    rs2::config conf;
    conf.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
    conf.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

    conf.enable_record_to_file("map.bag");

    // Post-processing filters
    rs2::decimation_filter dec_filter;
    rs2::spatial_filter spat_filter;
    // Configure filter parameters
    dec_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 3);
    spat_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.55f);

    rs2::pipeline_profile profile = pipe.start(conf);
    rs2_stream align_to = RS2_STREAM_COLOR;
    rs2::align align(align_to);
    bool die(false);
    while (!die) {
        rs2::frameset frames = pipe.wait_for_frames();
        auto aligned_frames = align.process(frames);

        rs2::video_frame post_processed_image_frame = aligned_frames.first(RS2_STREAM_COLOR);
        rs2::frame post_processed_depth_frame = aligned_frames.get_depth_frame();

        post_processed_depth_frame = dec_filter.process(post_processed_depth_frame);
        post_processed_depth_frame = spat_filter.process(post_processed_depth_frame);
         
        // Creating OpenCV Matrix from a depth image
        Mat depth(Size(640, 480), CV_16UC1, (void*)post_processed_depth_frame.get_data(), Mat::AUTO_STEP);

        // Creating OpenCV Matrix from a color image
        Mat color(Size(640, 480), CV_8UC3, (void*)post_processed_image_frame.get_data(), Mat::AUTO_STEP);

        cv::imwrite(dirColor + sTime + suffix, color);
        cv::imwrite(dirDepth + sTime + suffix,depth);
        }

Could it be because some values in the filtered depth image are converted to `Null`?

ev-mp, 2018-12-25:
@armandok , the decimation filter [modifies image size](https://github.com/IntelRealSense/librealsense/blob/development/doc/post-processing-filters.md#decimation-filter).
You need to query the filtered frame for its actual size, then apply it in code
>// Creating OpenCV Matrix from a depth image
    Mat depth(~~Size(640, 480)~~, CV_16UC1, (void*)post_processed_depth_frame.get_d

See similar report #2802.

RealSense-Customer-Engineering, 2019-01-08:
[Realsense Customer Engineering Team Comment]
@armandok Did you get through the issue as ev-mp suggested? Looking for your update. Thanks!

armandok, 2019-01-09:
Thank you so much @ev-mp. That solved my problem.
@RealSense-Customer-Engineering Yes!

closed 2019-01-09 by armandok

RealSense-Customer-Engineering, 2019-01-14:
[Realsense Customer Engineering Team Comment]
@armandok Any update about this issue? Looking forward to your reply. Thanks!

armandok, 2019-01-14:
@RealSense-Customer-Engineering  I just removed the decimation filter and it worked.