realsenseai/librealsense #2961: Compiler Error during Installation of librealsense Libraries

ShadowHunter0510, 2018-12-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version | Ubuntu 16 | 
| Platform                                 | Raspberry Pi |

### Issue Description
I am trying to connect an Intel RealSense D435 to my Raspberry Pi by following the instructions in [Linux Ubuntu Installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md), but because I've had no experience in programming the Raspberry Pi, I've run into a couple of problems that may be due to a fundamental misunderstanding of how coding works. 

I've made my way to the last line of code: `sudo make uninstall && make clean && make && sudo make install`. However (and this has occurred a couple of times before), I get an error such as the one in the pictures down below. Sometimes Ubuntu just crashes altogether. Another problem before the installation is that Ubuntu cannot find anything to uninstall (hence: "cannot find install manifest").
[image]
[image]

### Attempts

* Initially, I could not navigate to the librealsense directory. I solved this problem by cloning the librealsense files from GitHub.
* I tried typing in `git checkout v2.17.0` as a user previously advised before the make and cmake.
* I have bought a charger that supplies more current as the lightning bolt consistently shows up without it.
* I am now considering updating the build toolchain to gcc-5 as the error message includes a link about that and maybe creating a swap file in case the RAM capacity is not sufficient.

### Questions/Other Issues

* I also had problems when installing Intel RealSense permission scripts:
`sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/` 
`sudo udevadm control --reload-rules && udevadm trigger`
When I enter these two lines of code, it sometimes prompts me for my password, then it just moves to the next line as if I hadn't typed in anything then hit enter. Is something else supposed to happen?

* Because of the ambiguity of which directory I should be doing all of this, I've been using /home as my root until the instructions tell me to switch to the librealsense directory. In the final line of code, I've been using the /build branch located in librealsense.

* Finally, I've been skipping the steps that apply to only Ubuntu 14 because the version of my operating system is Ubuntu 16. I also skipped the steps regarding kernel modules (as stated in [Intel® RealSense™ D400 cameras with Raspberry Pi](https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md)) according to the directions. I don't know if this is the right procedure especially because when I start up my Raspberry Pi, the boot screen notifies that it has "failed to start Load Kernel Modules".

dorodnic, 2018-12-25:
Hi @ShadowHunter0510 

I think this will have a rather simple solution

In [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) it states:
> Note: If you encounter the following error during compilation gcc: internal compiler error it might indicate that you do not have enough memory or swap space on your machine. Try closing memory consuming applications, and if you are running inside a VM increase available RAM to at least 2 GB.

You can use the following guide to set-up a swap file - [how-to-add-swap-space-on-ubuntu-18-04](https://linuxize.com/post/how-to-add-swap-space-on-ubuntu-18-04/)

It's been a while since I built the SDK on a Pi, but I did run compilation on a ROC-RK3328-CC, and had the same exact problem, that was resolved after setting up a swap file.

As for udev rules and other steps, I think you are on the right track, please let us know if you encounter any problems. 

Patching can be an issue if the kernel is monolithic (which is often the case for single-board computers).
However, there is a decent chance the camera will work without patching. It depends on the kernel version, you can check with `uname -r`. Most 4.4s will offer streaming, 4.16 will also support hardware timestamps without patching. 
If you do encounter issues related to patching, please consider to add `-DFORCE_LIBUVC=true` to your CMake command. This mode uses somewhat different driver stack, that works better on some systems.

ShadowHunter0510, 2018-12-25:
Thank you so much for your prompt response, @dorodnic 

I will try to set up a swap file as indicated in the link. After entering `uname -r` into terminal, I can confirm that my kernel version is 4.19.8-v7+.

Upon digging around in my files, I also realized that a folder for librealsense was created in my home folder. Should I delete this folder before attempting installation again?

dorodnic, 2018-12-25:
I think 1GB should do it. If you run into the same problem - try increasing it again. Also (this helped me on some occasions) close Firefox / Chromium when building the SDK to save memory. 

There are also ways to cross-compile (build the SDK for the Pi on your main PC and copy the file to the Pi)

4.19 should require no patching for the D435. There has been some [reports of problems](https://github.com/IntelRealSense/librealsense/issues/2850) with the latest 4.19, so consider using the LIBUVC route (should work on any kernel)

You don't have to delete the other folder.

ShadowHunter0510, 2018-12-25:
I have an update. At around 42% (where it normally fails) some warnings pop up regarding ISO C++11 and macro ‘QW_TO_QUAD’. I’m not sure about the significance of this, but the complication continues to run despite these warnings.

[image]
[image]

ShadowHunter0510, 2018-12-26:
Hi @dorodnic 

The compiler successfully finished. I’m a bit confused on how to open the software, however. I tried plugging the Intel RealSense D435 Camera into one of my USB ports on my Raspberry Pi, but nothing comes up on the computer screen. Is an application supposed to automatically start or must I physically run and execute the Intel RealSense Viewer?