realsenseai/librealsense #2962: Error Upon Opening RealSense Viewer

ShadowHunter0510, 2018-12-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | v2.17.0 | 
| Operating System & Version | Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    | 4.19.8-v7+ | 
| Platform                                 | Raspberry Pi |

### Issue Description
After installing the SDK for the Intel RealSense Depth Camera, I opened the RealSense Viewer to use the camera. The viewer recognizes that a camera has been plugged into the USB port, but it outputs an error upon booting. From my understanding, it may have something to do with starting the device. A picture for reference can be seen down below.

[image]

dorodnic, 2018-12-26:
Hi @ShadowHunter0510 
Is the camera working correctly on another PC, perhaps on Windows?
One thing to get off the troubleshooting list is to run firmware update on a separate Windows PC. 
Usually this error has something to do with camera firmware / calibration

ShadowHunter0510, 2018-12-26:
Hey @dorodnic 

For clarification, do I update the firmware on the camera itself following these directions in this [link](
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf) or on the actual PC itself? I’ll try downloading the SDK for my Windows PC and see if the camera works there. I’m relatively new to all of this, so I’m sorry if I’m slow to understand.

dorodnic, 2018-12-26:
Hi @ShadowHunter0510 
I recommend to do the firmware update on a Windows PC.
We will do our best to assist with any question you might have.

ShadowHunter0510, 2018-12-26:
Hi @dorodnic 
I just downloaded the SDK onto my Windows PC and the Intel RealSense Depth Camera works fine on it. However, it still does not work on the Raspberry Pi through Ubuntu Mate. What troubleshooting steps can I follow?

ShadowHunter0510, 2018-12-26:
Hi @dorodnic 
Another update: After running the DFU tool and obtaining the .bin file that contains the latest firmware, I'm having trouble installing the firmware on the camera. Before updating the camera, I check that my camera is in the updatable devices list. Then, when I enter the path and filename, the camera disconnects from my computer by itself and says that no device was connected, even though it previously acknowledged that it recognized the presence of a camera. A picture can be found down below.

[image]

dorodnic, 2018-12-26:
Could you please check the camera firmware version? This can be done from the Viewer, by clicking on the (i) icon under device name.

ShadowHunter0510, 2018-12-26:
It's 05.09.02.00, as shown in the picture of my last comment as well.

ShadowHunter0510, 2018-12-26:
The name is Intel RealSense USB2. The details under "physical port" has some gibberish. Should Windows computers support USB 3? I know that the Raspberry Pi only supports USB 2.0.
[image]

dorodnic, 2018-12-26:
Thank you,
I believe firmware update is at the moment limited to x86 Windows / Ubuntu 16.04 and to USB3. 
I think firmware update can make a significant difference (hope it will resolve the issue on the Pi), because USB2 support was officially added **after** 5.9.2

ShadowHunter0510, 2018-12-26:
Thank you @dorodnic 
Should I now try to update the firmware through Ubuntu instead because my Windows PC doesn't support USB 3.0?

ShadowHunter0510, 2018-12-26:
Hi @dorodnic 

After installing the firmware on another computer, I am pleased to say that the camera is functioning on my Raspberry Pi! Thank you for your help!

dorodnic, 2018-12-26:
Hi @ShadowHunter0510 

Glad it helped!

The reason other examples don't work is because of USB2. In general, USB3 is **strongly** recommended if you want to use RealSense depth-cameras, since the bandwidth from multiple streams easily saturate USB2 capacity. 
USB2 is partially supported by the firmware specifically to enable platforms like the Raspberry Pi, but you don't really get 100% of camera functionality this way.

I'd recommend considering something like [ROC-RK3328-CC](http://en.t-firefly.com/product/rocrk3328cc.html) which is slightly more powerful ARM single-board computer, or the [Up-Board](https://up-shop.org/27-up) which has Raspberry Pi form factor and Intel CPU. Both support USB3.

IMHO the best way to get started is with [python](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python). 
We have fairly detailed tutorial on detecting objects with estimated distance - 
[Distance to Object](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb)
You can even try live it in the browser by clicking on [image]
If you decide to go with C++, [rs-dnn](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/dnn) opencv example is very similar.

ShadowHunter0510, 2018-12-26:
Hey @dorodnic 

Thanks for all of your suggestions! I've been checking out the ROC-RK3328-CC and I'm definitely considering buying it because of its similar functionalities. I noticed that it has a TF card reader instead of a MicroSD card reader. Since all of my files, including the Intel RealSense SDK and Ubuntu Mate, are located on the SD card, I would not have to reinstall everything, correct? I believe that the MicroSD card can be inserted in the reader.

dorodnic, 2018-12-27:
I believe the board can boot from a MicroSD card (at least the one I tested could).
Not sure if it is compatible with the Pi image, I never tried it.

ShadowHunter0510, 2018-12-27:
Hi @dorodnic 

I have another question that I haven’t been able to figure out. Once I write some code for the Intel RealSense Camera and compile it, how do I run it live? What I would feasibly like to do is take current live video from the camera, analyze the scene with the code, and have the Raspberry Pi send some output. Since I’m not particularly well-versed in this kind of software to hardware translation, I’m a bit confused.

dorodnic, 2018-12-28:
Hi @ShadowHunter0510 
Perhaps its best to open a new issue since the original problem is now resolved.
In python example the line `cfg.enable_device_from_file("../object_detection.bag")` makes it run from previously collected recording. If you remove it, the demo will try to connect to a live camera. 
Now you can just put everything starting from `frameset = pipe.wait_for_frames()` is a while-loop and once you get a bounding box and the distance, send them over the network for example (or log to file, or anything else). This completely depends on your application and interests. You could play with OpenCV Kalman_filter to tracked detected objects in the scene, use depth to cut only the detected object, you could use depth to make sure detected object is real, and not just a flat picture, etc...

closed 2019-01-10 by RealSense-Customer-Engineering