realsenseai/librealsense #2963: error when running the test file

Buted, 2018-12-26:
Hello!
I have installed the librealsense on Ubuntu 18.04 LTS. But when I ran the test program (copy the README file's program), one problem occurred to me.
when I imported pyrealsense2.so, two warnings happened:
> test.py:2: FutureWarning: pybind11-bound class 'pyrealsense2.processing_block' is using an old-style placement-new '__init__' which has been deprecated. See the upgrade guide in pybind11's docs. This message is only visible when compiled in debug mode.
  import pyrealsense2 as rs

>test.py:2: FutureWarning: pybind11-bound class 'pyrealsense2.filter' is using an old-style placement-new '__init__' which has been deprecated. See the upgrade guide in pybind11's docs. This message is only visible when compiled in debug mode.
  import pyrealsense2 as rs

Then I ignore them, continue running the test file. A new error happened:
>    Traceback (most recent call last):
  File "test.py", line 7, in <module>
    pipeline.start()
RuntimeError: No device connected

The ubuntu system is installed on the VMware Workstation Pro 15, but it told me that the carema is connected.
Now, I'm confused. Can I use the carema on the virtual machine? Or have I made a wrong installation?

dorodnic, 2018-12-26:
Hi @Buted 
You can use the camera with VMWare virtual machine. 
Did you install librealsense from apt-get or built from source?
What do you get when you call `lsusb` from terminal?
What about `rs-enumerate-devices`?

Buted, 2018-12-26:
@dorodnic 
I built librealsense from source.
When I call **lsusb**:
>Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 010: ID 041e:409b Creative Technology, Ltd 
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

The second is the camera.

When I call **rs-enumerate-devices**:
>rs-enumerate-devices: command not found

Does it mean that I failed to build the librealsense?

Buted, 2018-12-29:
@dorodnic 
That works for me!
Thanks a lot!

closed 2018-12-29 by Buted