realsenseai/librealsense #2966: feature request: release the gil on wait_for_frames

migel, 2018-12-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.17.0                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
Is it possible the release the gil when calling `pipline.wait_for_frames()` (like in `frame_queue.wait_for_frame`)?

dorodnic, 2018-12-26:
Hi @migel 
I think that makes sense. Sure, will do. Also, feel free to make the modifications you want directly and [contribute](https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md) the change.

closed 2019-01-10 by RealSense-Customer-Engineering

landersson, 2019-04-02:
Did this get fixed? I just ran into this as well using pyrealsense2-2.19.2.692, which makes my python GUI app pretty much unusable.

landersson, 2019-04-02:
This one line change seem to fix the problem for me:

https://github.com/landersson/librealsense/commit/5e49019eec8648698b7b03f5472e7f690db88274

As long as pipeline::wait_for_frames() doesn't call any python callbacks or otherwise interacts with the python interpreter, I guess this should be possible to merge?