realsenseai/librealsense #2967: Win10 running and installing the SDK example dnn, how is the direct exit

honghande, 2018-12-27:
My installation environment is:

1. The system is win10.

2.rssetup-2.17.0.394.exe

3.VS2015

honghande, 2018-12-28:
@ev-mp 

Thank you. The Caffe files are all under the path. The main reason is that my camera doesn't have a color camera.

RealSense-Customer-Engineering, 2019-01-02:
[Realsense Customer Engineering Team Comment]
@Honghande So this ticket can be closed, right?

honghande, 2019-01-03:
@RealSense-Customer-Engineering 

You can't understand what you're saying.

RealSense-Customer-Engineering, 2019-01-03:
[Realsense Customer Engineering Team Comment]
@honghande As you said, the issue was caused by no color camera. Please try the example with the camera with color sensor like D415/D435. Or you need modify the code which will be tracked in https://github.com/IntelRealSense/librealsense/issues/2972

closed 2019-01-08 by RealSense-Customer-Engineering