realsenseai/librealsense #2968: CMake Compiled source code appears：error: could not find git for clone of libusb

honghande, 2018-12-27:
1. Computer Networking

2.CMake3.13

3. Download：https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv

ev-mp, 2018-12-27:
@honghande , 
Building librealsense with defaults requires `git` being installed. This is one of the `BUILD_WITH_TM2` 
 CMake's option prerequisites.
You can either install `git`, or unselect that option in order to proceed.
See #2719  for details.

honghande, 2018-12-28:
Thank you, yes, I installed git, now I can make Cmake, but I also tried not to choose BUILD_WITH_TM2 is still invalid, so the best way is to install git.

honghande, 2018-12-28:
@ev-mp Thank you, yes, I installed git, now I can make Cmake, but I also tried not to choose BUILD_WITH_TM2 is still invalid, so the best way is to install git.

closed 2019-01-03 by RealSense-Customer-Engineering