realsenseai/librealsense #2969: Offline Depth Image Filtering

anupamsobti, 2018-12-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Operating System & Version |   Ubuntu 16.04  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy + 2 }                          | 
| Language                            |  {C++/OpenCV }                          | 
| Segment			|  Others }                          | 

### Issue Description
I have a dataset which I collected using R200. I saved the depth images as 16-bit png files. I was trying to use the filters available in the latest API for post-processing the depth images. 
The hurdle I face is the conversion from Mat to rs2::frame. Is there a way to read 16-bit png files to rs2::frame (instead of OpenCV imread). I saw that the conversion is supported other way round, i.e., rs2::frame --> Mat but couldn't find the conversion Mat --> rs2::frame. 
In case this conversion isn't possible, can I use the filters on Mat directly?

dorodnic, 2018-12-27:
Hi @anupamsobti 
[software-device](https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device) example show how you could generate new `rs2::frame` objects. Please note that filters may need some additional information, such as depth-units, baseline, and possibly intrinsics.

anupamsobti, 2018-12-27:
Thanks for the prompt reply. Would you recommend trying out the filters using dummy information with the real depth image data? Or are there fixed values I could use for R200?

closed 2019-01-15 by RealSense-Customer-Engineering