realsenseai/librealsense #2972: Can color images of infrared cameras replace color images of color cameras?

honghande, 2018-12-27:
librealsense/notebooks/distance_to_object.ipynb   Can I use this demo for color infrared images? 
Because my depth camera is based on D405, only use two infrared cameras, no color cameras, but infrared cameras are with color. In addition, when running DNN in sdk, because there is no color camera flashing back, can color infrared image replace color camera image?

dorodnic, 2018-12-27:
Hi @honghande 
D405 can stand for several different things, could you please share the Product ID as reported in RealSense Viewer?
I'll assume the device is similar in its function to D410 (which in turn is D415 without RGB)
To get RGB instead of Y8 from the infrared imager, you need to add:
```cpp
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, RS2_FORMAT_RGB8);
pipe.start(cfg);
```
Also, I believe visual preset 6 is "Remove IR pattern" that reduces the laser in infrared RGB image.
As far as I remember, the SDK should be robust to this kind of configuration, meaning that `frameset::get_color_frame()` should return infrared RGB if proper RGB isn't available.

honghande, 2018-12-28:
@dorodnic Thank you very much. I chose to use infrared color image instead of color camera image. Now there is still a problem. I want to use Intel Real Sense Viewer to collect infrared color image and depth image for color point cloud display. But the effect of point cloud display is not good. Is the data collected wrong, because I take other people's image is very good?

The following are demo color maps, depth maps and point clouds.
[image]
[image]
[image]
The following are Intel's infrared color maps, depth maps and point clouds.
[image]
[image]
[image]

RealSense-Customer-Engineering, 2019-01-03:
[Realsense Customer Engineering Team Comment]
Hi @honghande

Thank you for the RealSense Viewer screenshot. The RealSense Viewer should be able to stream both depth and Infrared 1.
Just want to be sure, does the modified code explicitly enable the depth stream?

    cfg.enable_stream(RS2_STREAM_DEPTH, RS2_FORMAT_Z16);
    cfg.enable_stream(RS2_STREAM_INFRARED, RS2_FORMAT_RGB8);

Regarding to the depth quality issue, has the problem been resolved?

If you have NDA in place with Intel, please log on to https://intel.zendesk.com/hc/en-us/requests/new?ticket_form_id=589348
It is RealSense support portal for NDA customers.

Thank you.

honghande, 2019-01-03:
hi,@RealSense-Customer-Engineering ,

I want to use the above camera to run the DNN example in Intel's demo: The problem now is to modify it and still exit directly.

honghande, 2019-01-03:
Hi, @RealSense-Customer-Engineering When you ask questions on the Intel website, you feel not very used and convenient. All kinds of validation and experience are not very good.

honghande, 2019-01-03:
Hi @dorodnic @RealSense-Customer-Engineering 
My modified code is here:
https://github.com/honghande/intel_DNN

RealSenseCustomerSupport, 2019-01-31:
----------------------------------------------

XiangyuX Sun, Jan 30, 11:47 PM

@honghande For CNN example, I can get your code working with SDK 2.18 and D415.  Could you please check if you can run the example with the same setting?

RealSenseCustomerSupport, 2019-02-11:
----------------------------------------------

@honghande Any other questions about this ticket?

dorodnic, 2019-02-17:
@honghande @sherrysun2018 - consider using `Remove IR pattern` preset to enhance RGB from IR image.

closed 2019-02-22 by RealSenseCustomerSupport