realsenseai/librealsense #2977: how can i compile pcl-color?

hfutzhu, 2018-12-28:
I clone  the librealsense and then  add    'add_subdirectory(pcl-color)'    at  the end of  the wrappers/pcl/cmakelist.txt, but CMake Error at wrappers/pcl/pcl-color/CMakeLists.txt:22 (find_package)
the pcl-color/cmakelist.txt:
set( realsense2_DIR "C:/Program Files/librealsense2/lib/cmake/realsense2" CACHE PATH "Path to librealsense2 config directory." )

how can i set the realsense2_DIR?

hfutzhu, 2018-12-28:
error:
CMake Error at wrappers/pcl/pcl-color/CMakeLists.txt:22 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

RealSense-Customer-Engineering, 2019-01-04:
[Realsense Customer Engineering Team Comment]
Hi @hfutzhu,

Do you use Ubuntu or Win10? Please follow the instructions in below link.
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl

RealSense-Customer-Engineering, 2019-01-07:
[Realsense Customer Engineering Team Comment]
Hi @hfutzhu,

You can check below modification in order to build the rs-pcl-color sample app.
https://github.com/freemanlo/librealsense/tree/issue2977/wrappers/pcl

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

Hi @hfutzhu,

I will close this if no further question.

closed 2019-02-19 by RealSenseCustomerSupport