realsenseai/librealsense #2979: Cant disable fisheye autoexposure.

rshanor, 2018-12-28:
|Required Info | |
|----------------------------------------|------------------------------------------- 
| Camera Model                       | D400 |
| Firmware Version                   | 05.10.11.90 and 05.09.13.99 |
| Operating System & Version  |   Linux (Ubuntu 18) |
| Kernel Version (Linux Only)    |  4.15.0  |                                  
| Platform                                  | Laptop |
| SDK Version                           |  2.17     |                     
| Language                               |  C++ |

### Issue Description

Setting the RS2_OPTION_ENABLE_AUTO_EXPOSURE option quietly fails for the fisheye sensor. sensor.set_option does not throw and error, and sensor.get_option returns the expected value, whatever I last set. However, frame metadata RS2_FRAME_METADATA_AUTO_EXPOSURE always returns 1,  even after setting RS2_OPTION_ENABLE_AUTO_EXPOSURE to 0. Also values I set to RS2_OPTION_EXPOSURE get overwritten, even though documentation states "Setting any value will disable auto exposure".

This parameter behaves as expected on the INFRARED stream. 

It looks like the realsense viewer does the right thing, but the parameters on the device look the same as compared to my test script. 

### Pseudocode
```                    
sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0.0f);
for (auto&& frame : pipe.wait_for_frames()) {
    if (frame.get_profile().stream_type() == RS2_STREAM_INFRARED) {
        frame.get_frame_metadata(RS2_FRAME_METADATA_AUTO_EXPOSURE) // Equals 1 always
    }
}
sensor.get_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE); // Equals 0
```

dorodnic, 2018-12-29:
Hi @rshanor 
D425 I assume from the description, is this correct? Specifically, I'm looking for the Product ID as reported in the Viewer, so we can reproduce the problem

rshanor, 2019-01-02:
Sorry for the delay. Intel RealSense D430 model, Product ID 0AD5.

rshanor, 2019-01-02:
Hmm so I think I see the issue. I was using the ctx->query_devices()[0].query_sensors() to disable auto exposure but using a rs2::pipeline to acquire frames. This creates two different auto exposure objects and causes disagreement between settings. Not sure if this is a bug or a feature...

dorodnic, 2019-01-03:
Hi @rshanor 
Sorry for the delay. I was trying to reproduce this yesterday and ran into some unrelated problems with my D425 device. 
To be honest, I'm not sure if this is a bug or a feature either. Every call to `query_devices` performs new enumeration and we don't keep the list of previously discovered devices. This makes thread-safety of the whole thing much easier, but any device state that is software-only, such as presets and fisheye auto-exposure, is duplicated.
The way we recommend to overcome this is by getting the device via `auto profile = pipe.start(); profile.get_device();` 
I think it would be a good idea to test "stateful context" feature on a side branch, and try to integrate it over time.

rshanor, 2019-01-03:
Thanks for the follow up @dorodnic , that all makes sense.  Now that I actually understand how everything works, no issue from my end.

closed 2019-01-03 by rshanor