realsenseai/librealsense #2981: rs-depth example, how to read the depth values of each pixel?

abdelrahmanhussein, 2018-12-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D435 | 
| Firmware Version                   | 5.10.13 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-42-generic |                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.17.0                         | 
| Language                            |  C                          | 
| Segment			|  others                           | 

### Issue Description

Hello,

When I tried to rin rs-depth example, which is supposed to return depth value of each pixel, I ran the example using the following command to stream the output to output.txt file.

> rs-depth > output.txt

 This image show the part of the content in the output.txt file: https://ibb.co/q16v7wT

The describtion of this example demonstrates that it prints text-based representation of the depth map. 

This is the code that carries out this conversion to text-based representation:
```
/* Print a simple text-based representation of the image, by breaking it into 10x5 pixel regions and approximating the coverage of pixels within one meter */
            out = buffer;
            int coverage[ROW_LENGTH] = { 0 }, x, y, i;
            for (y = 0; y < HEIGHT; ++y)
            {
                for (x = 0; x < WIDTH; ++x)
                {
                    // Create a depth histogram to each row
                    int coverage_index = x / WIDTH_RATIO;
                    int depth = *depth_frame_data++;
                    if (depth > 0 && depth < one_meter)
                        ++coverage[coverage_index];
                }

                if ((y % HEIGHT_RATIO) == (HEIGHT_RATIO-1))
                {
                    for (i = 0; i < (ROW_LENGTH); ++i)
                    {
                        static const char* pixels = " .:nhBXWW";
                        int pixel_index = (coverage[i] / (HEIGHT_RATIO * WIDTH_RATIO / sizeof(pixels)));
                        *out++ = pixels[pixel_index];
                        coverage[i] = 0;
                    }
                    *out++ = '\n';
                }
            }
            *out++ = 0;
            printf("\n%s", buffer);

```

So my questions are:

1) How to modify this code to make it produce a conventional depth map (by that I mean the distance of each pixel of the frame from the camera)?

2) Is there some documentation that demonstrates the libraries used to operate the camera, so that I would be able to use their functions inside my own code?

Thanks

HippoEug, 2019-01-07:
My project involves creating a depth map of the 1280x720 picture using C++. I simply recorded a .bag file, and then using the SDK's convert tool to convert that .bag file into a .csv file with 921,600 data points. Hope this helps in some way shape or form!

closed 2019-01-10 by abdelrahmanhussein

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
@abdelrahmanhussein Any other questions about this ticket? Looking forward to your update. Thanks!

abdelrahmanhussein, 2019-01-11:
No, thank you so much for your wonderful support. It is working fine with me now.

RealSense-Customer-Engineering, 2019-01-15:
[Realsense Customer Engineering Team Comment]
@abdelrahmanhussein If no other questions for this ticket, will close the ticket accordingly. Could you please update? Thanks!

abdelrahmanhussein, 2019-01-15:
@RealSense-Customer-Engineering Isn't it already closed? I closed it 6 days ago.

RealSense-Customer-Engineering, 2019-01-16:
[Realsense Customer Engineering Team Comment]
@abdelrahmanhussein Sorry that we're having some problem to sync github to our internal support platform, so I didn't notice your reply.  Now I see you closed it. :)