realsenseai/librealsense #2982: Unable to compute aligned cloud from Color Frame

hrdkjain, 2018-12-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |   2.17.0                          | 
| Language                            |  C++                          | 
| Segment			|  others                         | 

### Issue Description
I am capturing a frameset and exporting it to ply. Then I am using the next frameset to recreate the captured ply from color Frame by:
1. Projecting color pixel to depth pixel using `rs2_project_color_pixel_to_depth_pixel`

1. 

I want to recreate cloud from the color Frame for my application, here is the code that I used:

``

ev-mp, 2018-12-30:
Hello @hrdkjain , several comments:
>rs2_project_color_pixel_to_depth_pixel(depth_pixel,
        reinterpret_cast<const uint16_t*>(capturedFrameset.get_data()),

A frameset does not guarantee the first frame to be a depth frame - use explicit assignment to depth.

Verify that the rgb indexes are not swapped:
>  // recreate complete cloud from image
  for (int r=15; r<ColorFrame.get_height()-15; r++)  {
    for (int c=15; c<ColorFrame.get_width()-15; c++)  {
      float rgb_pixel[2],depth_pixel[2];
      rgb_pixel[0] = float(**C**);
      rgb_pixel[1] = float(**R**);

The video frame data alignment is always raw-major, hence rgb_pixel(i,j) refers to (row,column) .

I don't have a specific knowledge of how the Object file format works, but does the hard-coded value accounts for the 15 pixels margin that you impose in the repro?
>off_FS << 862500 << " 0 0\n"

Also note:
-Pointcloud generates a color coordinate for each depth pixel
-recomputeCloud(...) calculates depth coordinate for each color pixel
These are not the same

hrdkjain, 2019-01-05:
Thanks for your response 

> > rs2_project_color_pixel_to_depth_pixel(depth_pixel,
> > reinterpret_cast<const uint16_t*>(capturedFrameset.get_data()),
> 
> A frameset does not guarantee the first frame to be a depth frame - use explicit assignment to depth.

I think you mean this modification:
`reinterpret_cast<const uint16_t*>capturedFrameset.first(RS2_STREAM_DEPTH).get_data())`
> Verify that the rgb indexes are not swapped:
>
>
> The video frame data alignment is always raw-major, hence rgb_pixel(i,j) refers to (row,column) .

I reverified it, using 

> rgb_pixel(i,j) refers to (row,column) 

maps the transposed image.
> 
> I don't have a specific knowledge of how the Object file format works, but does the hard-coded value accounts for the 15 pixels margin that you impose in the repro?
> 
> > off_FS << 862500 << " 0 0\n"
> 
Well that's not a smart way of doing it, but this value is calculated before hand and accounts for the 15 pixel margin.
> Also note:
> -Pointcloud generates a color coordinate for each depth pixel
> -recomputeCloud(...) calculates depth coordinate for each color pixel
> These are not the same

That's an interesting highlight, meaning that both the methods might not be able to actually give the same number of points or exactly same location of points, however don't you think that the surfaces created by both the methods should be similar. 
Looking at the image 3, its hard for me to imagine that the two surfaces would be similar but rather seems to shifted wrt to one another. Could you shed some light on that, if thats bound to happen or if there is something wrong in the `recomputeCloud(...)`

My aim is to detect the markers in the rgb image and locate them precisely in the 3D point cloud.

ev-mp, 2019-01-08:
On the last point:
 - The depth and RGB sensors are mostly overlap, but due to the horizontal displacement the rightest depth pixel columns are not in RGB's FOV. Hence when mapping depth to color, the most left columns of color pixels will not be utilized and vice versa. 
- The number of points generated is a function of the input frame resolution only and is not affected by the displacement.
 - The theoretic limit of Depth->RGB mapping precision is 0.5 pixel due to discretization of continuous (0..1) texture coordinates*.  Which means that the full circle of Depth(i,j)->RGB(m,n)->Depth(x,y) mappings may certainly result in (i,j) and (x,y) being within 1 pixel radius.
- On top of it you should add that the Depth and RGB calibrations exhibit intrinsically a certain error tolerance, which will add up for the displacements.

With all that's said, the image (4) in the above post produces quite consistent correlation, which suggests that your device is properly calibrated.

RealSenseCustomerSupport, 2019-02-12:
----------------------------------------------

Hi @hrdkjain,

Did you resolve the issue per @ev-mp's comment?

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi @hrdkjain,

If no further question, will close this.

closed 2019-03-15 by hrdkjain