realsenseai/librealsense #2992: Change in the shape of point cloud captured from python API

Pratiksha2692, 2019-01-02:
Hi, 
Here is camera and os info  

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435| 
| Firmware Version                   | (05.10.06.00) | 
| Operating System & Version |   Linux (Ubuntu 16.04) |                                     | 
| Language                            |  python                        |                       

### Issue Description
I am using Realsense camera D435 for 3d reconstruction of object. I have aligned color and depth image using API. Using blue color extraction opencv algorithm, I am able to get only box object and white markers.From this depth image I have created point cloud(x, y, z data- .npy file). I have converted this .npy file into .ply file using Open3d.  When I opened this .ply in Meshlab, it is showing that the point cloud does not preserve its shape as compared to color image. The ply file is disturbed and overall shape is changed. It became bulge at center.  What is the issue?

 Color image is as below.

[image]

The ply file is as below.
[image]

Can you please help me out? 
Thanks in advance.

RealSense-Customer-Engineering, 2019-01-09:
[Realsense Customer Engineering Team Comment]
Hello @Pratisksha2692,

Thank you for your question.

You are setting your objects on a glass/Plexiglas plane, there have been reports in the past about RealSense Cameras having difficulties detecting objects and determining accurate depth ranges on objects behind reflective surfaces.

Would it be possible to remove the glass/Plexiglas plane from your project and attempt to gather images again?  I would like to see if the .ply file will have a bulge in the center.

Thank you!

Pratiksha2692, 2019-01-11:
Thanks for the reply.
The ply image which I attached above is taken on the wooden platform.

RealSenseCustomerSupport, 2019-01-18:
Hello @Pratishksha2692,

Sorry for the delay, its possible that your camera might need to be calibrated.

Before you calibrate, be sure to upgrade your camera firmware to latest FW:

-  Download latest FW here (5.11.1):  https://downloadcenter.intel.com/download/28481/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?product=128255

-  Upgrade latest FW with Linux DFU:  https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf

-  After firmware is updated, please follow the Dynamic Calibration user guide located here:  https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

-  You will need to print out a target for camera to be calibrated with:
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/print-target-fixed-width.pdf

Please update this ticket with your results.

Thank you!

RealSenseCustomerSupport, 2019-01-29:
----------------------------------------------

Nate Ward, Jan 29, 11:15 AM

No response in 5 days.

Close this issue.

closed 2019-02-05 by RealSenseCustomerSupport