realsenseai/librealsense #2995: Openni wrapper build documentaition

rlsutton1, 2019-01-02:
The documentation for the openni wrapper does not explain nor could I find documented any where how to get the build to compile the openni wrapper driver. Can this information be added to the documentation?

Having to compile 2 SDK's is quite a barrier for anyone who is not a C developer and given that openni has not changed in 5 years, would it be possible to make a build of the driver lib for the verious OS/hardware combinations available?

MartyG-RealSense, 2019-01-02:
Are you looking for the source for OpenNI2?

https://github.com/occipital/openni2

If not, binaries are also available.

https://structure.io/openni

rlsutton1, 2019-01-03:
Thanks MartyG but no I have already built the wrapper and have my realsense device working with openni.

I was hoping the process could be better documented and just easier.

I might add that my build of the driver involved manually editing the make and copying src files from openni as well as copying some realsense libs and src around. So all up a real mess - but I have it working.

closed 2019-01-07 by RealSense-Customer-Engineering