realsenseai/librealsense #2996: IMU sample in Unity

mtschoen, 2019-01-02:
Hello there,

I'm sure this is being worked on, but is there an available sample using IMU data in Unity? I'd like to try a simple experiment where the point cloud is counter-rotated by IMU data to stabilize the horizon. I'm a little confused about where to start. I think I want to add a POSE profile to my stream reader, but I'm unsure what kind of format/dimensions this "frame" should be.

Thanks!

MartyG-RealSense, 2019-01-02:
As you observed, there is not a Unity sample program for IMU data yet.  A RealSense community member has contributed a non-Unity program for rotating the point cloud with the IMU though.  Looking at this code may provide some useful insights.

https://github.com/GruffyPuffy/imutest

mtschoen, 2019-01-02:
Thanks for the quick reply. I am currently looking through samples and actually encountered this one already. It seems that the Unity wrapper has a slightly different way of accessing frame data than the C++ examples, so I'm still not entirely sure how to proceed. I will keep searching, and if/when I have a solution I will post here.

ogoshen, 2019-01-03:
The C# wrapper doesn't yet contain the high-level APIs to support IMU, but I can point you in the right direction for low-level access.

dorodnic, 2019-01-03:
We are working on better IMU support across the SDK. 

The plans is:
a. Motion and Pose frames support in C++ and python (**Done**)
b. IMU integration with existing C++ examples  (**Done**)
c. Adding new C++ examples with IMU (**WIP**)
d. Adjust the .NET wrapper and add to Unity example

mtschoen-unity, 2019-01-04:
Thanks for the update.

mtschoen, 2019-01-04:
Thanks for the update. I probably won't have time to try @ogoshen's suggestion, but thanks for that, too! :)

closed 2019-01-14 by RealSense-Customer-Engineering