realsenseai/librealsense #3000: where is the rs2driver.dll file? I cannot see rs2driver.dll...please help me

SeokHawn, 2019-01-04:
Hello, I want to use OpenNI2 with realsense 415.
I saw realsense's github and did as written on realsense github.
But I could not find rs2driver.dll.
May I ask you where is the rs2driver.dll file?
(Even though rs2driver.dll is made by debugging visual studio rs2driver.sln,  OpenNI2 examples is seemed that could not find device)

cucrui, 2019-01-09:
> Hello, I want to use OpenNI2 with realsense 415.
> I saw realsense's github and did as written on realsense github.
> But I could not find rs2driver.dll.
> May I ask you where is the rs2driver.dll file? [...]

same problem as you, but my system is Linux, I cant find out the file "librs2driver.so" either.

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
@SeokHawn  Please refer to https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2#getting-started
You need run Cmake, then generate project files and compile driver to get rs2driver.dll.
Also you need copy rs2driver.dll and realsense2.dll to OPENNI2_DIR/Samples/Bin/OpenNI2/Drivers/

RealSense-Customer-Engineering, 2019-01-15:
[Realsense Customer Engineering Team Comment]
@SeokHawn Did you get through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-01-18:
@SeokHawn Any update for this issue? Thanks!

SeokHawn, 2019-01-18:
thanks! everyone.
Finally, I got the file!

closed 2019-01-18 by SeokHawn