realsenseai/librealsense #3001: GPIO pin for inter-device communication

zuru, 2019-01-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |  5.9.14 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.15                          | 
| Language                            |  C++                          | 
| Segment			|  VR/AR                        | 

### Issue Description
Hello, trying to locate the documentation regarding the GPIO pins i noticed that it has been deleted in a previous commit (https://github.com/IntelRealSense/librealsense/commit/28371bc6de865f10b54265f808cdcd8f50bdbe07).

The past version can be found at: https://github.com/IntelRealSense/librealsense/blob/28371bc6de865f10b54265f808cdcd8f50bdbe07/doc/rs400/external_devices.md

Is this still supported? According to this issue #2512, it is not.

We need this to be able to send a signal on demand from one of our D415's to other D415s via the GPIO pin in order to establish global time correspondences when recording from a set of HW-synced devices connected via LAN.

MartyG-RealSense, 2019-01-04:
Page 47 of the November 2018 edition of the RealSense 400 Series data sheet document contains a table of pin details.

[image]

The data sheet can be downloaded from the link below:

https://www.intel.co.uk/content/www/uk/en/support/articles/000026827/emerging-technologies/intel-realsense-technology.html

One of the Intel guys (I do not work for Intel) can best inform you about the extent to which pin access is supported in the camera's firmware driver.

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
I just tried on D415.
Unfortunately, the GPIOs control seems not open to users due to device firmware being locked. Only unlocked device (for engineering) is programmable on the GPIOs.

closed 2019-02-17 by dorodnic