realsenseai/librealsense #3002: Background extraction from the object in python

Pratiksha2692, 2019-01-04:
Hi,
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (05.10.06.00) | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {3d reconstruction}                          | 

### Issue Description
I have to extract the object from its background. I had put the realsense camera at the corner of platform. RGB image is attached as below.

[image]

I have to do 3d reconstruction of model. I am using opencv_pointcloud_viewer.py code to capture point cloud.  How can I extract model from the background in the point cloud? How to use minimum and maximum depth_clamp value in the code? 

Kindly advise.

Thanks in advance.

Pratiksha2692, 2019-01-09:
Any suggestion from Realsense engineering team?

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

Hello @Pratiksha2692,

Sorry for the delay.

Have you taken a look at the documentation of API?

https://intelrealsense.github.io/librealsense/doxygen/pages.html

Also here is some info regarding OpenCV background subtraction.

https://docs.opencv.org/3.4/db/d5c/tutorial_py_bg_subtraction.html

Please take a look at these docs and update ticket with your results.

Thank you!

Pratiksha2692, 2019-03-05:
Hi @RealSenseCustomerSupport ,

Thanks a lot. 

I had tried OpenCV background subtraction. It worked fine for rgb image.

`self.color_image_1 = np.asanyarray(self.color_frame_1.get_data())` 
which has the shape (480,640,3).
I want to apply same background extraction on the .ply file. But while capturing the .ply file I had done some advance setting regarding depth_table to get particular area of interest. 

`depth_table_control_group.depthClampMin = 0
  depth_table_control_group.depthClampMax = 800`
So the x,y,z,r,g,b data points are less than (640*480).
How can I apply this background extraction on .ply file?

Thanks in advance.

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi Pratiksha2692,

Wonder if you have any updates/progress on this one?

Python wrapper newly had threshold filter added in, which is similar to what RS Viewer has in post processing filter section.
You can set min_distance and max_distance to do depth segmentation. Below is an example of using the filter.

a) First claim a filter. which by default uses 0.15~4m): thr_filter = rs.threshold_filter()
b) And then call set_option to adjust min&max distance value (float type):
thr_filter.set_option(rs.option.min_distance, <YOUR_MIN_VALUE>)
thr_filter.set_option(rs.option.max_distance, <YOUR_MAX_VALUE>)

And then from there, you will need to your own way to deal with the data from PLY file.
Hope this can be helpful.

Thanks!

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi Pratiksha2692,

Do you still need help? If nothing else is needed, it will be closed.

Thanks!

closed 2019-03-30 by RealSenseCustomerSupport