realsenseai/librealsense #3003: undefined reference using model-views.h

Wh1teSh4dow, 2019-01-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      |
| Firmware Version                   | 05.10.13.00                                            | 
| Operating System & Version | Ubuntu 16.04 LTS                                   | 
| Kernel Version (Linux Only)    |  4.7.0.intel.r5.0                                         | 
| Platform                                 | Skull Canyon  |
| SDK Version                            |  2.17.0                          | 
| Language                            |  C++                          | 
| Segment			| Automotive                          | 

Hi all,

I'm trying to use the model-views library which seems to be providing quite some good functions for handling multiple camera and their configuration.

However, trying to simply include the file alone needs to some issues.

Having only the below minimal `testfile.cpp`:

    #include "model-views.h"

    int main()
    {
        return 0;
    }

building and linking as below:
`g++ -std=c++11 -o rs-modelviews -I/home/alex/realsense/librealsense/librealsense-2.14.0/librealsense-2.17.0/common -I/home/alex/realsense/librealsense/librealsense-2.14.0/librealsense-2.17.0/third-party/imgui -I/home/alex/realsense/librealsense/librealsense-2.14.0/librealsense-2.17.0/src -L/usr/local/lib rs_modelviews.cpp -lrealsense2 && ./rs-modelviews`

I get the following error:

    /tmp/ccHgupyo.o: In function `__static_initialization_and_destruction_0(int, int)':
    rs_modelviews.cpp:(.text+0x266c): undefined reference to `from_rgba(unsigned char, unsigned char, unsigned char, unsig
    rs_modelviews.cpp:(.text+0x26a3): undefined reference to `from_rgba(unsigned char, unsigned char, unsigned char, unsig
    rs_modelviews.cpp:(.text+0x26da): undefined reference to `from_rgba(unsigned char, unsigned char, unsigned char, unsig
    rs_modelviews.cpp:(.text+0x2711): undefined reference to `from_rgba(unsigned char, unsigned char, unsigned char, unsig
    rs_modelviews.cpp:(.text+0x2748): undefined reference to `from_rgba(unsigned char, unsigned char, unsigned char, unsig
    /tmp/ccHgupyo.o:rs_modelviews.cpp:(.text+0x277f): more undefined references to `from_rgba(unsigned char, unsigned char
    collect2: error: ld returned 1 exit status

From here I'm not quite sure what I can try. the "from_rgba" functions listed are defined in the model_views.h file included. I can't seem to find a shared or static library created from the model-views so I don't think I'm missing anything on that side.

I'm also trying to avoid using any cmake here as my ultimate workflow must be done using a makefile so I can't copy the cmake used to create the realsense-viewer and the makefile generated is way beyond my level of expertise in makefile creation.

I have created and run multiple software using the librealsense without any issues but this is the first time I'm attempting to use the model-views.

Any hint would be more than welcome.

ev-mp, 2019-01-07:
The header "model-views.h" is utilized in SDK's demos - it is not part of the public API.
Pls note that significant share of the functionality provided is explicitly reliant on third-party packages, and cherry-picking could be troublesome. 

Except for those constrains  you may use it as any other standard C++ class/utility.

The implementations you failed to link in are contained in "model-views.**cpp**"
https://github.com/IntelRealSense/librealsense/blob/development/common/model-views.cpp#L52

Wh1teSh4dow, 2019-01-08:
> The header "model-views.h" is utilized in SDK's demos - it is not part of the public API.
> Pls note that significant share of the functionality provided is explicitly reliant on third-party packages, [...]

Ok, it's good to know that the library isn't part of the public API.

Exactly, the implementation is in the cpp file and the definition of the function is provided in the header file: https://github.com/IntelRealSense/librealsense/blob/development/common/model-views.h#L38

Are you suggesting that I include the cpp file too? When there's a header file, I can't recall having had to include the cpp file as well, in the past.

dorodnic, 2019-01-09:
Hi @Wh1teSh4dow 
Well, all code that resides outside the `include` and `src` folders is provided as a sample code, and not being  compiled into `realsense2.so/dll` library. This means that you can't use it as a library (include just h and lib/a), but you can freely copy paste any part of it into your application (under very permissive [Apache 2.0](https://github.com/IntelRealSense/librealsense/blob/master/LICENSE) license)
So in case of `model-views` you are encouraged to pick the parts you need and use them directly in your application. 
In general, `model-views` specifically **requires** severe refactoring on our part, and is not representative of the code inside the library.

Wh1teSh4dow, 2019-01-09:
Okay, so what I'm hearing is I can use any of the content of those files but it'll be challenging to include the files as-is, is that what you mean?

dorodnic, 2019-01-14:
Yes, that's correct. Its basically reference code, and can be subject to breaking changes. 
What exactly are you trying to do, I'd love to try to help?
For multi-camera, we have [multicam](https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam) example

Wh1teSh4dow, 2019-01-14:
@dorodnic 
Thanks for trying to help. I'm trying to create a software component which can pull RGB, IR and D streams from multiple realsense cameras without using poll_for_frames, rather wait_for_frames. 
Obviously, I can't use wait_for_frames for multiple devices sequentially so my idea was to create a multithreaded component where each thread would "wait for frames" for each realsense camera.

For now, whether this works or not is only speculation on my side, but that at least is my idea. As you said, multicam was my reference and it relied on the library mentioned in this thread which is why I wanted to use it, but later on I realized what I wrote here, which is that it works with "poll_for_frames" so it's not so crucial anymore to use that specific library as there's more things I'll need to recode than I initially hoped.

closed 2019-01-17 by Wh1teSh4dow