realsenseai/librealsense #3004: Box Dimensioner Multicam,Run demo and report errors :

honghande, 2019-01-05:
Traceback (most recent call last):
  File "E:/intelRealsense/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py", line 143, in <module>
    run_demo()
  File "E:/intelRealsense/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py", line 72, in run_demo
    transformation_result_kabsch  = pose_estimator.perform_pose_estimation()
  File "E:\intelRealsense\python\examples\box_dimensioner_multicam\calibration_kabsch.py", line 198, in perform_pose_estimation
    corners3D = self.get_chessboard_corners_in3d()
  File "E:\intelRealsense\python\examples\box_dimensioner_multicam\calibration_kabsch.py", line 164, in get_chessboard_corners_in3d
    depth = get_depth_at_pixel(depth_frame, corner[0], corner[1])
  File "E:\intelRealsense\python\examples\box_dimensioner_multicam\helper_functions.py", line 117, in get_depth_at_pixel
    return depth_frame.as_depth_frame().get_distance(round(pixel_x), round(pixel_y))
TypeError: get_distance(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.depth_frame, x: int, y: int) -> float

Invoked with: <pyrealsense2.depth_frame object at 0x0000000009C33260>, 259.0, 23.0

honghande, 2019-01-05:
I opened the checkerboard photo, ran the code, and the camera pointed at the photo,

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
Hello @honghande,

Do you use D415 or D435? And, which librealsense version used?

SchotmanG, 2019-01-14:
Required Info | -
--- | --- 
Camera Model | D435
Firmware version | 5.11.1
OS | Ubuntu 16.04
Kernel version | 4.15
realsense version | latest from github

Similar problem:

``` 
Traceback (most recent call last):
  File "box_dimensioner_multicam_demo.py", line 143, in <module>
    run_demo()
  File "box_dimensioner_multicam_demo.py", line 52, in run_demo
    frames = device_manager.poll_frames()
  File "/home/marc/Documents/workspace/point_clouds/examples/box_dimensioner_multicam/realsense_device_manager.py", line 211, in poll_frames
    device.pipeline.poll_for_frames(frameset)
TypeError: poll_for_frames(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pipeline) -> pyrealsense2.composite_frame

Invoked with: <pyrealsense2.pipeline object at 0x7fa12f64fc38>, <pyrealsense2.composite_frame object at 0x7fa12f64fe30>
```

I tried changing the poll_frames function as suggested here: https://github.com/IntelRealSense/librealsense/pull/2949/commits/44ac0c508fd8452005dee61e05c51ed15570045b
No succes...

RealSenseCustomerSupport, 2019-01-22:
Hello @SchotmanG,

With the same fix applied on the script in librealsense 2.17.1, I don't observe the issue you mentioned. I didn't print the chessboard and just used my another LCD screen to display the pattern picture for D435 camera.
The calibration process can be still done successfully.

closed 2019-03-04 by RealSenseCustomerSupport