realsenseai/librealsense #3007: UE4 Wrapper Project

smudgy67, 2019-01-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Firmware Version                   | (Open RealSense Viewer --> 3.21.0.0) | 
| Operating System & Version |   { Win 10 }  | 
| Kernel Version (Linux Only)    |  (N/A)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { Intel RealSense SDK 2.0 }                          | 
| Language                            |  {unrealengine4}                          | 
| Segment			|  {wrapper project }                          | 

### Issue Description
Hey there,

I have followed the setup instructions as per https://github.com/IntelRealSense/librealsense/tree/development/wrappers/unrealengine4, however, on opening the UE4 project, RealSenseUE, and hitting play, nothing happens visually.
The SR300 webcam activates, green light on, though the boxes remain in standard grid material.

These are the details of my webcam:
[image]

Please can you assist? Thank you.

Kind regards,

Dominic

wongfei, 2019-01-07:
@smudgy67 
please post log file located in: "ProjectDir\Saved\Logs\RealSenseUE.log"

smudgy67, 2019-01-08:
Hey wongfei,

Please find attached the log file as requested.

Kind regards,

Dominic

smudgy67, 2019-01-08:
[RealSenseUE.log](https://github.com/IntelRealSense/librealsense/files/2736197/RealSenseUE.log)
Hope this helps wongfei?!
And thank you for your help too =]

wongfei, 2019-01-08:
@smudgy67 

seems you running modified project because it tries to load a preset and records to file:
```
[2019.01.07-11.15.39:948][854]LogRealSense: Error: rs2_serialize_json failed: Object does not support "librealsense::ds5_advanced_mode_interface" interface! 
[2019.01.07-11.15.39:948][854]LogRealSense: Error: Read failed: test_preset.json Size=0
[2019.01.07-11.15.39:949][854]LogRealSense: Display: enable_record_to_file: test_record.bin
```

check BP_RealSenseInspector options:
[image]

check BP_RealSenseAPI:
[image]

smudgy67, 2019-01-09:
Hey wongfei,

For the BP_RealSenseInspector, I've tried all the options, however, no joy.

In relation to the BP_RealSenseAPI, selecting the 'RunRealSenseDemo', does nothing. Though when driving through the 'ExampleCXollection' node, the camera turns on for the 10 seconds.

Most bizarre Dude!?!

[rsopt]<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F13559668%2F50851082-6d303f00-1384-11e9-9730-e697ffbb2408.jpg&data=02%7C01%7C%7C8571e5b5360e493f7a2308d67597e5ca%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636825692581044119&sdata=4AuF%2FYw7jvPRsaOi%2BRmL5rg0QLc4fCfRiInEfOt0bI8%3D&reserved=0>

[rsapi]<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F13559668%2F50850975-22162c00-1384-11e9-8c75-941bdcabd3c2.jpg&data=02%7C01%7C%7C8571e5b5360e493f7a2308d67597e5ca%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636825692581044119&sdata=emxI21DQ2Hnp2rBXzM6KC9apg8mh35svfqWmbxPJGJs%3D&reserved=0>

wongfei, 2019-01-09:
@smudgy67 
revert all custom changes in plugin
in BP_RealSenseInspector disable all options except DepthRaw
set depth to 640x480@30
try running

in case it doesn't help, edit file:
[ProjectName/Plugins/RealSense/Source/RealSense/Private/RealSenseInspector.cpp]

LINE: 432
REALSENSE_TRACE(TEXT("Frameset %d"), FramesetId); // <--- uncomment
FramesetId++;

build solution for "Development Editor" configuration and try running
upload the log

dorodnic, 2019-01-14:
Hi @smudgy67 
The Unreal wrapper was contributed supporting only D400 (was tested with D415 and D435). The change to enable SR300 should be a minor one, looking at the code, the methods [SavePreset and LoadPreset](https://github.com/IntelRealSense/librealsense/blob/a321a0404c7b0cfd2e157d85c6b00f88a3e55d48/wrappers/unrealengine4/Plugins/RealSense/Source/RealSense/Private/RealSenseDevice.cpp#L124-L199) will not work on the SR300. You could replace them with `return true;`.

smudgy67, 2019-01-14:
Hey Sergey,

I tried this as instructed:
[cid:32fe5b40-546a-4275-b705-162b8f0dd319]

However, on play, just this appears:
[cid:a501725a-996e-4fce-a6b5-2935e113cd2b]

The camera comes on for around 10s and then turns itself off.
In addition, the camera will only turn on with the following connected:
[cid:9c96d84d-865c-4115-87fd-5a03215c0aec]

Please find attached a copy of the project logfile.

Thank you for your assistance.

Kind regards,

Dominic

dorodnic, 2019-01-15:
Hi @smudgy67 
Could you please re-upload attachments, they got removed by GitHub.

smudgy67, 2019-01-15:
Hey you go Sergey...
[RealSenseUE.log](https://github.com/IntelRealSense/librealsense/files/2760178/RealSenseUE.log)

smudgy67, 2019-01-15:
I tried this as instructed:
[image]

However, on play, just this appears:
[image]

The camera comes on for around 10s and then turns itself off.
In addition, the camera will only turn on with the following connected:
[image]

Please find attached a copy of the project logfile.

Thank you for your assistance.

Kind regards,

Dominic

smudgy67, 2019-03-19:
I updated the SR300 camera to the D435 and all seems to be working ok, thank you.

closed 2019-03-19 by smudgy67

RealSenseCustomerSupport, 2019-03-19:
----------------------------------------------

Hi smudgy67,

Glad to hear that you have a solution and can move forward on this one.
Again sorry for taking a while to check on you.

Thanks!

smudgy67, 2019-03-19:
There's another open issue, that is stills outstanding. Setting up a D435i camera that I purchased as an interim camera, prior to the D435.
I still have issues with that one.