realsenseai/librealsense #3011: How to Use Python to Reduce the Error of Depth Value

chaojizhangdaxin, 2019-01-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 14/16/17)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.16.1 }                          | 
| Language                            |  {opencv/python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
My project is to identify objects within 5 meters, and I found that the farther away from the camera, the greater the error of the depth value obtained.How to use Python to reduce the error of depth value?

RealSenseCustomerSupport, 2019-01-18:
Hi @chaojizhangdaxin,

You can review below BKM to adjust the parameters. Changing the preset and laser power is a quick way to have a different result.
https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

RealSenseCustomerSupport, 2019-02-17:
----------------------------------------------

Hi @chaojizhangdaxin,

Still need any support for this topic?

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

Also refer to some suggestions in #3139.

closed 2019-05-27 by RealSenseCustomerSupport

viswaroopam, 2019-08-27:
The links like (https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf) are broken. Is it possible to restore/redirect these links to the active page?