realsenseai/librealsense #3012: Can't connect D435i!!!!!

AdamPengG, 2019-01-07:
I have install the realsense sdk librealsense-2.17.1 and ROS wrapper realsense-2.1.3.And I follow the https://github.com/intel-ros/realsense/wiki/SLAM-with-D435i and install all the components.
Finally I run the roslaunch realsense2_camera opensource_tracking.launch.But the terminal shows I did't connect the the D435i.

`[ERROR] [1546838519.355105483]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1546838519.355165653]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1546838519.355291096]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager
[camera/realsense2_camera-3] process has died [pid 10090, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/peng/.ros/log/27cdb684-123c-11e9-8909-107b447e20dc/camera-realsense2_camera-3.log].
log file: /home/peng/.ros/log/27cdb684-123c-11e9-8909-107b447e20dc/camera-realsense2_camera-3*.log
[ WARN] [1546838525.358489472]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/color/image_raw,
   /camera/aligned_depth_to_color/image_raw,
   /camera/color/camera_info

'
`What should i do?

RealSenseCustomerSupport, 2019-01-19:
Hello @AdamPengG,

Sorry to hear you are having issues with D435i.

-  Are you able to run any of the SDK samples you have installed?  Samples such as "realsense-viewer", or "rs-capture"?  I would like to verify that the camera is working properly.

-  If the SDK apps do not work, then something with your setup may be affecting the camera's connection to your system.  You can try the SDK on a Windows system as well to verify camera works.

-  Ensure you are plugging in camera to USB 3.0 ports.

-  Also, be sure to upgrade your camera firmware to latest FW 5.11.1:

- Download latest FW here (5.11.1): https://downloadcenter.intel.com/download/28481/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?product=128255

- Upgrade latest FW with Linux DFU: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf

Please update this ticket with your results.
Thank you!

RealSenseCustomerSupport, 2019-01-25:
No response in over a week.
Close issue.

closed 2019-02-01 by RealSenseCustomerSupport